UbiquityRobotics/raspicam_node #22: Support request for usb webcams

adnion, 2017-06-18:
In need for a high update rate I search for ways to quickly transfer images with RPi's wifi.
The package "video_stream_opencv" doesn't offer compression and "usb_cam" is outdated and throws many errors.
Would it be possible to use the Pi's hardware encoder for arbitrary webcams?

davecrawley, 2017-06-18:
At least in principal yes. RPi cams can do high frame rate and fast shutter speed. Currently its 1 cam per 1 Raspberry pi, but you could network together several RPis if you needed more than 1.

You need to spend time looking at the RPi video modes, as there are trade offs between speed, frame rate, and resolution. Ultimately these things are all limited by how quickly the camera can download frames from the camera to the RPi over the MIPI interface. As the MIPI interface is a lot faster than USB 2.0 it usually works much better on a RPi than it does on a typical webcam.

jim-v, 2017-07-22:
The raspicam_node is specific to the Raspberry Pi camera.  There are several ROS nodes for USB cameras,
for example [http://wiki.ros.org/uvc_camera]

closed 2017-07-22 by jim-v