UbiquityRobotics/raspicam_node #23: Performance issues with raspicam_node

surfertas, 2017-09-06:
Hi, thanks a lot for contributing raspicam_node. Just out of curiosity what kind of publishing rates do you experience with this node running on a Raspberry Pi 3 Module. I currently have it running with a frequency of ~3.8 hz. Was wondering if this was the expected publishing rate. What kind of publishing rate are you experiencing. 

The current set up I have is:
Master machine: Ubuntu 16.04.2 LTS
Raspi: Ubuntu 16.04.2MATE

rohbotics, 2017-09-06:
Thanks for opening the issue.

Some more information would be helpful would be:

- How did you install raspicam_node (from source, debs, etc)?
- What command are you using to launch it?
- What is the output of `uname -a` on your system?

surfertas, 2017-09-06:
1. I followed the instructions here: https://github.com/UbiquityRobotics/raspicam_node
2. Using this command to launch: roslaunch camerav2_1280x960.launch 
3. 
Raspi: Linux user_name 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l armv7l armv7l GNU/Linux

Master: Linux user_name 4.4.0-78-generic #99-Ubuntu SMP Thu Apr 27 15:29:09 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

rohbotics, 2017-09-06:
Thanks.

Some things to try:
Try recompiling with `catkin_make -DCMAKE_BUILD_TYPE=RelWithDebInfo` in your `~/catkin_ws`.

What is the rate of the `camera_info` topic, is it also slow?

Can you also try running the master, raspicam_node, and `rostopic hz /raspicam_node/image/compressed` all on the Pi, with `ROS_MASTER_URI=http://localhost:11311`. This should isolate you from network issues.

I will try to reproduce your issues if I can. Just to confirm is your image the one from here: https://ubuntu-mate.org/download/ or here https://ubuntu-pi-flavour-maker.org/download/ ?

surfertas, 2017-09-06:
This is the results I get when running everything on the PI. I get a publishing rate of ~30hz.

subscribed to [/raspicam_node/image/compressed]
average rate: 30.035
	min: 0.031s max: 0.036s std dev: 0.00141s window: 29

subscribed to [/raspicam_node/camera_info]
average rate: 30.005
	min: 0.030s max: 0.036s std dev: 0.00132s window: 29

surfertas, 2017-09-06:
I get this over my network using raspicam_node:
subscribed to [/raspicam_node/image/compressed]

average rate: 5.957
	min: 0.160s max: 0.180s std dev: 0.00743s window: 5

Using video_stream_opencv package I get below fwiw over my network:
subscribed to [/webcam/image_raw/compressed]
average rate: 30.746
	min: 0.020s max: 0.044s std dev: 0.00578s window: 27

rohbotics, 2017-09-07:
I haven't gotten to reproducing the issue yet, but this is another thing that you could try

`rosrun image_transport republish compressed in:=/raspicam_node/image compressed out:=/image/image` on the pi, along with raspicam_node.

and check the rate of `/image/image/compressed` on your computer.

surfertas, 2017-09-07:
subscribed to [/raspicam_node/image/compressed]
average rate: 5.249
	min: 0.187s max: 0.196s std dev: 0.00400s window: 4

and after the using "republish"
subscribed to [/image/image/compressed]
average rate: 3.290
	min: 0.302s max: 0.305s std dev: 0.00158s window: 3

Can I understand the design reason not to use image_transport in this case?

rohbotics, 2017-09-07:
`image_transport` was not used because the Raspberry Pi image processing pipeline can take advantage of the hardware JPEG compression in the GPU, and `image_transport` can only take raw image data and compress it directly. If there was a way to provide the compressed data directly to image_transport, we would use it.

surfertas, 2017-09-07:
Just to confirm that you are getting 30hz over the network using raspicam_node?

rohbotics, 2017-09-07:
I just tested it, and I am getting ~10-15hz instead of the expected 30 when running over the network, I am getting 30hz on the pi.

rohbotics, 2017-09-07:
I set the resolution lower, down to 640x480 and I am getting the expected 30hz. This is looking more like a network performance issue... 

What are the results of iperf between your computer and the pi?
Run `iperf -s` on the pi and `iperf -c {PI_IP}` on your computer.

I am getting:
```
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec  63.1 MBytes  52.7 Mbits/sec
```

surfertas, 2017-09-08:
Ok. confirmed I am getting much better with reduced dimensions. 640x480. I get ~25hz. Fwiw, the video_stream_opencv was pumping through images of dimensions 640x480, so we werent comparing apples to apples to earlier. Maybe worth mentioning that if expecting a decent publishing rate over the network need to set with smaller dimensions, as there will be disappointment when they see a rate of 5-6hz. Thanks for taking the time to answer the questions!!!

rohbotics, 2017-09-09:
I added a note about publish rate over the network in the troubleshooting section of the README.

If you feel this issue is resolved, go ahead and close it.

Thank you for using raspicam_node, and good luck on whatever you build with it! If you encounter any more problems, don't hesitate to open another issue on this repo.

rohbotics, 2017-09-23:
@surfertas 

I am closing this issue now. Let us know if you have any further problems with raspicam_node via github.

closed 2017-09-23 by rohbotics