UbiquityRobotics/raspicam_node #28: Ubuntu-Mate (16.04.3) Source Compile error

red24dog, 2017-10-08:
Raspberry Pi 3 (Turtlebot3) w/V2 camera, Linux cassini 4.9.53-v7+ #1040 SMP Fri Oct 6 14:19:18 BST 2017 armv7l armv7l armv7l GNU/Linux, 
following https://github.com/UbiquityRobotics/raspicam_node/issues/new instructions.

Having to reinstall Ubuntu-Mate/ROS and raspicam_node (due SD Card crash) in my RasPi3, resulting in many compile errors in the "RaspiCamControl.h" . Thus the "make -j4 -l4" failed. Appreciate your advice. Ross

[ 84%] Built target turtlebot3_msgs_generate_messages
[ 86%] Generating dynamic reconfigure files from cfg/Camera.cfg: /home/turtlebot/catkin_ws/devel/include/raspicam_node/CameraConfig.h /home/turtlebot/catkin_ws/devel/lib/python2.7/dist-packages/raspicam_node/cfg/CameraConfig.py
Scanning dependencies of target raspicli
Scanning dependencies of target raspicamcontrol
[ 88%] Building CXX object raspicam_node/CMakeFiles/raspicli.dir/src/RaspiCLI.cpp.o
[ 90%] Building CXX object raspicam_node/CMakeFiles/raspicamcontrol.dir/src/RaspiCamControl.cpp.o
Generating reconfiguration files for Camera in raspicam_node
Wrote header file in /home/turtlebot/catkin_ws/devel/include/raspicam_node/CameraConfig.h
[ 90%] Built target raspicam_node_gencfg
In file included from /home/turtlebot/catkin_ws/src/raspicam_node/src/RaspiCamControl.cpp:37:0:
/home/turtlebot/catkin_ws/src/raspicam_node/include/RaspiCamControl.h:116:5: error: ‘MMAL_PARAM_EXPOSUREMODE_T’ does not name a type
     MMAL_PARAM_EXPOSUREMODE_T exposureMode;
     ^
/home/turtlebot/catkin_ws/src/raspicam_node/include/RaspiCamControl.h:117:5: error: ‘MMAL_PARAM_EXPOSUREMETERINGMODE_T’ does not name a type
     MMAL_PARAM_EXPOSUREMETERINGMODE_T exposureMeterMode;. 

and many more similar errors . The RaspiCamControl,h file is in the include directory

rohbotics, 2017-10-08:
What is the output of `apt policy libraspberrypi0` and `apt policy libraspberrypi-dev`?

rohbotics, 2017-10-08:
@red24dog I looked into it some more, and looks like a bug that slipped in during some code cleanup.

Can you try the `bugfix-#28` branch by running `git checkout bugfix-#28` and trying to compile it again?

Thanks for reporting the issue.

rohbotics, 2017-10-08:
Thanks for the additional details.

For the bugfix, run `git checkout bugfix-#28` in `~/catkin_ws/src/raspicam_node`, then `cd ~/catkin_ws` and then try `catkin_make` again.

Let me know if it works.

red24dog, 2017-10-08:
Fixed the issue. I figured out that I needed to remove the current raspicam_node directory from the /src directory, then clone a new code that contained the bug fix. I appreciate your really fast response! I have a demo planned with some folks tomorrow so glad the Turtlebot 3 works well again.

rohbotics, 2017-10-08:
@red24dog Thanks for reporting this issue, I have merged in the fix.

Good luck with your demo, and anything else you build using raspicam_node!

PR #29 (2017-10-08): add proper includes to header RaspiCamControl
Fixes #28
closed 2017-10-08 by rohbotics