UbiquityRobotics/raspicam_node #30: Turtlebot3 RaspiCam/Android-Teleop No Image

red24dog, 2017-10-09:
Issue (as originally submitted Oct 4/17 on discourse.ros.org): Trying to view the Raspicam Publication “/raspicam_node/image/compressed” on connected Galaxy Tab ( 4.0.4) Android Teleop but with no image is displayed (though Teleop works). 

Action: 1) Inquiring “rosnode info /android/camera_view” that node appears to use a Subscription to “/compressed_image [unknown type]”. 
2) I tried a suggested workaround "rosrun image_transport republish compressed in:=/raspicam_node/image raw out:=/raspicam_node/image". Attached "rqt_graph" report shows successful new image nodes and Android connections but still no android image. [TB3-Wii-Android-rosgraph_201771009.png.pdf](https://github.com/UbiquityRobotics/raspicam_node/files/1369516/TB3-Wii-Android-rosgraph_201771009.png.pdf). Appreciate any troubleshooting actions - Thanks.

rohbotics, 2017-10-10:
As the republish is running, can you have the android app connected to the uncompressed stream at /raspicam_node/image. 

I don't have any experience with the Android app you are talking about, is it open source/publicly available? Do you have a link?

red24dog, 2017-10-10:
No, if the republish is running , the image is not displayed on the Android Tab, though the Teleop works.

 

The App, https://play.google.com/store/apps/details?id=com.github.rosjava.android_apps.teleop.indigo <https://play.google.com/store/apps/details?id=com.github.rosjava.android_apps.teleop.indigo&hl=en> &hl=en 

 

 

http://www.ros.org/wiki/android_teleop 

 

https://github.com/ros-android/android_teleop 

 

I’ll

davecrawley, 2017-10-10:
> rosjava.android_apps.teleop.indigo <https://play.google.com/
> UbiquityRobotics/raspicam_node/issues/30#issuecomment-335329872> , or
> AAzx8JVXF2gMR0CgXgEp9FB06MuyUrthks5sqsK7gaJpZM4Py5zB> . <

Before we get too off topic. We have an application that runs on android
that does teleoperation like things, called robot commander, but it also
allows for voice control of the robot. Its here:

https://play.google.com/store/apps/details?id=com.jrlandau.robotcmdr&hl=en

Currently, its tested against our standard image. Voice control includes
commands like "go forward" but if you have some kind of localization and
navigation functioning you can also do things like "set waypoint 'kitchen'
 " and drive the robot around and then tell the robot to "go to 'kitchen'
 " and it will go there. On our standard image, it functions with our
navigation node "move basic".

Links are as follows:

Video of robot commander working:
https://www.facebook.com/ubiquityrobotics/videos/1506195869418429/
link to our standard image for Raspberry Pi 3 that contains recent -
although not the latest - versions of all our software in a single coherent
set.
https://downloads.ubiquityrobotics.com/
Link to our move_basic github repo
https://github.com/UbiquityRobotics/move_basic

David

rohbotics, 2017-10-10:
@davecrawley I think robot commander is a slightly different use case, but we should consider adapting it to solve the same problem.

@red24dog Looks like your app is subscribing on a different image topic, try this republish `rosrun image_transport republish compressed in:=/raspicam_node/image compressed out:=/image_raw`.

red24dog, 2017-10-11:
on roscore 
Trial 1:machine: rosrun image_transport republish compressed in:=/raspicam_node/image compressed out:=/image_raw. No image on android. rqt_image_view shows a choice of >raspicam_node/image/compressed and /image_raw/compressed both displaying and image.

turtlebot@cassini:~$ rosnode list
/android/camera_view
/android/virtual_joystick
/dashboard
/image_republisher_1507687046708849557
/masterNameResolver
/raspicam_node
/rosout
/rqt_gui_cpp_node_2675
/turtlebot3_core
/turtlebot3_lds
turtlebot@cassini:~$ rosnode info /android/camera_view
--------------------------------------------------------------------------------
Node [/android/camera_view]
Publications: 
 * /rosout [rosgraph_msgs/Log]

Subscriptions: 
 * /compressed_image [unknown type]

Services: None

contacting node http://192.168.2.29:58248/ ...
Pid: 4800
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound
    * transport: TCPROS

Trial 2: trying compressed_throttle and compressed_image, neither description exists.

turtlebot@cassini:~$ rosrun image_transport republish compressed in:=/raspicam_node/image compressed out:=/image_raw  [Displays image on rescore machine]
turtlebot@cassini:~$ 
turtlebot@cassini:~$ rosrun image_transport republish compressed in:=/raspicam_node/image compressed_throttle out:=/image_raw
terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
  what():  According to the loaded plugin descriptions the class image_transport/compressed_throttle_pub with base class type image_transport::PublisherPlugin does not exist. Declared types are  image_transport/compressed_pub image_transport/raw_pub
Aborted
turtlebot@cassini:~$ rosrun image_transport republish compressed in:=/raspicam_node/image compressed_image out:=/image_raw 
terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
  what():  According to the loaded plugin descriptions the class image_transport/compressed_image_pub with base class type image_transport::PublisherPlugin does not exist. Declared types are  image_transport/compressed_pub image_transport/raw_pub

A curious observation is that the Subscribed Camera topic is compressed_throttle, on line 68-72  of the https://github.com/ros-android/android_teleop/blob/master/src/ros/android/teleop/Teleop.java,   

if (getIntent().hasExtra("camera_topic")) {
      cameraTopic = getIntent().getStringExtra("camera_topic");
    } else {
      cameraTopic = "camera/rgb/image_color/compressed_throttle";
    }

Yet, the Subscription appears to be different , /compressed_image
turtlebot@herschel:~$ rosnode info /android/camera_view
--------------------------------------------------------------------------------
Node [/android/camera_view]
Publications: 
 * /rosout [rosgraph_msgs/Log]

Subscriptions: 
 * /compressed_image [unknown type]

Services: None

I have to add that I am having a similar problem with my Turtlebot/Indigo/Kinect. The Android teleop worked ok until about 3 years ago and then stopped. Perhaps there's been a change in something. Anyway, I want to focus on the TB3/raspicam.

rohbotics, 2017-10-11:
@red24dog I edited your comment for formatting.

Try 
`rosrun image_transport republish compressed in:=/raspicam_node/image compressed out:=/compressed_image`

The current version of the app is here: https://github.com/rosjava/android_apps/tree/kinetic/teleop

red24dog, 2017-10-11:
Thanks for then Kinetic teleop repository, which I loaded into Android Studio sucessffuly and will attempt to install on my Android Tab.

Running your revised republish executed ok but still no connection between /android/camer_view and /compressed_image/compressed and no Android Tab image. Its should work as the /android/camera_view and /raspicam msg names now match.

turtlebot@herschel:~$ rosnode info /android/camera_view
--------------------------------------------------------------------------------
Node [/android/camera_view]
Publications: 
 * /rosout [rosgraph_msgs/Log]

Subscriptions: 
 * /compressed_image [unknown type]

Services: None

contacting node http://192.168.2.29:45409/ ...
Pid: 4557
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound
    * transport: TCPROS

[TB3-Android-compressed_20171010.png.pdf](https://github.com/UbiquityRobotics/raspicam_node/files/1377295/TB3-Android-compressed_20171010.png.pdf)

I'll advise results of loading the Kinetic Teleop App
Ross

red24dog, 2017-10-12:
I tried the compiled & installed Teleop-Kinetic app on my Android Tab - no joy - though the node info shows same info as my previous posting. If its useful , here's the image_republisher info

turtlebot@herschel:~$ rosnode info /image_republisher_1507764805619549533
--------------------------------------------------------------------------------
Node [/image_republisher_1507764805619549533]
Publications: 
 * /image_republisher_1507764805619549533/compressed/parameter_updates [dynamic_reconfigure/Config]
 * /image_republisher_1507764805619549533/compressed/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /compressed_image/compressed/parameter_updates [dynamic_reconfigure/Config]
 * /compressed_image/compressed/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /rosout [rosgraph_msgs/Log]
 * /compressed_image/compressed [sensor_msgs/CompressedImage]

Subscriptions: 
 * /raspicam_node/image/compressed [sensor_msgs/CompressedImage]

Services: 
 * /image_republisher_1507764805619549533/get_loggers
 * /image_republisher_1507764805619549533/compressed/set_parameters
 * /compressed_image/compressed/set_parameters
 * /image_republisher_1507764805619549533/set_logger_level

contacting node http://herschel.local:44233/ ...
Pid: 7218
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound
    * transport: TCPROS
 * topic: /compressed_image/compressed
    * to: /rqt_gui_cpp_node_7274
    * direction: outbound
    * transport: TCPROS
 * topic: /raspicam_node/image/compressed
    * to: /raspicam_node (http://cassini.local:39381/)
    * direction: inbound
    * transport: TCPROS

The only anomaly I note is the $rosnode info /android/camera_view Subscription lists /compressed_image [unknown type] 

rqt_reconfigure lists the republish type as jpeg, which I presume is correct.

Would by useful to post a ticket on the github.com rosjava/android repository?

rohbotics, 2017-12-17:
@red24dog

If you are still having problems, I would suggest a ticket on the repo that has the app, or a question on ros answers. 

I don't think there is anything we can do on the raspicam_node side to make this work for you.

Good luck!

closed 2017-12-17 by rohbotics