UbiquityRobotics/raspicam_node #32: We should do a release soon

jim-v, 2017-11-18:


rohbotics, 2017-11-19:
Try the deb here. https://packages.ubiquityrobotics.com/ubuntu/ubiquity-testing/pool/main/r/ros-kinetic-raspicam-node/ros-kinetic-raspicam-node_0.2.2-0xenial_armhf.deb 

You can install it with `dpkg -i`.

If it is good, I will sync to the main repo.

jim-v, 2017-11-19:
It looks good:  fiducials are detected when `roslaunch magni_nav aruco.launch` is run, and `roslaunch magni_nav aruco.launch` works.

```$ apt-cache policy ros-kinetic-raspicam-node 
ros-kinetic-raspicam-node:
  Installed: 0.2.2-0xenial
  Candidate: 0.2.2-0xenial
  Version table:
 *** 0.2.2-0xenial 100
        100 /var/lib/dpkg/status
     0.2.1-0xenial 500
        500 https://packages.ubiquityrobotics.com/ubuntu/ubiquity xenial/main armhf Packages
```

closed 2017-11-21 by jim-v