UbiquityRobotics/raspicam_node #34: support architecture 'i386' problem

thanhndv212, 2017-11-29:
I tried sudo apt-get update. I got this message. 

N: Skipping acquire of configured file 'main/binary-i386/Packages' as repository 'https://packages.ubiquityrobotics.com/ubuntu/ubiquity xenial InRelease' doesn't support architecture 'i386'

How can I overcome this issue? (maybe a version of amd64????)

rohbotics, 2017-11-29:
We don't support i386 on our repositories, only amd64 and armhf.

Additionally, raspicam_node is only available for armhf, as it does not actually work on anything other than a Raspberry Pi 2 or 3.

I may be able to help you more if I understood why you are trying to install raspicam_node on an something other than a pi?

davecrawley, 2017-11-30:
Yes its important to realize that this is a node for the Raspberry Pi
Camera, which attaches only to the Raspberry Pi set of computer boards. The
Raspberry Pi uses an ARM processor, so its not going to make any sense for
us to build this for anything else. As a practical point we only build for
the latest raspberry pis the Raspberry pi 2 and 3.

If you want to make use of a Raspberry Pi camera in your application
consider:

- Getting a raspberry pi 3 and attaching your camera to it
- Downloading our raspberry pi software image at
downloads.ubiquityrobotics.com and putting this image on the raspberry pi
- Connecting the raspberry pi via ethernet to your main computer which
could be an i386 machine

Play away and have fun

David

closed 2017-12-26 by rohbotics