UbiquityRobotics/raspicam_node #35: cannot locate rosdep definition for [libraspberrypi0]

dajiangsuo, 2017-12-08:
I followed the instructions and got the error when running "rosdep install --from-paths src --ignore-src --rosdistro=kinetic -y"

Here is what I did, any clues?
1. I am using Indigo rarther than kinetic
2. I created a new folder "/etc/ros/rosdep/sources.list.d/" and set up a file "30-ubiquity.list" and add this line to it: yaml https://raw.githubusercontent.com/UbiquityRobotics/rosdep/master/raspberry-pi.yaml
3. install dependencies using 
cd ~/catkin_ws
rosdep install --from-paths src --ignore-src --rosdistro=kinetic -y

then got the error above.

bg424, 2017-12-14:
did you updated rosdep (rosdep update) after setting up 30-ubiquity.list?

dajiangsuo, 2017-12-17:
I ran the update but got another one as to the camera_info_manager: 
raspicam_node: No definition of [camera_info_manager] for OS [debian]

rohbotics, 2017-12-17:
Looks like you missed the last step of ROS installation, run `sudo rosdep init` and then `rosdep update`. 

That should download all the dependency keys.

dajiangsuo, 2017-12-17:
Still doesn't work. Again, I am running ROS on raspberry pi and with Indigo version. Can this be an issue? Thanks

rohbotics, 2017-12-17:
What is the output of  `ls /etc/ros/rosdep/sources.list.d/`?

What about `cat /etc/ros/rosdep/sources.list.d/20-default.list`?

dajiangsuo, 2017-12-17:
Got "20-default.list  30-ubiquity.list"
**And**
# os-specific listings first
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml osx

# generic
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
gbpdistro https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml fuerte

# newer distributions (Groovy, Hydro, ...) must not be listed anymore, they are being fetched from the rosdistro index.yaml instead

rohbotics, 2017-12-17:
I edited you comment to make the formatting more readable.

Looking into it some more, looks like debian armhf was not supported by ROS Indigo. How did you install ros?

dajiangsuo, 2017-12-17:
I am using a hardware platform called "Navio2" on top of raspberrypi and the ros indigo comes with it

rohbotics, 2017-12-20:
I think that ROS indigo on debian is the main issue here.

You could try skipping this step and installing all the dependencies manually, but that may be a lot of effort. 

Does the OS you are on allow you to install ros packages via apt-get? Like `sudo apt-get install ros-indigo-camera-info-manager`?

AdriannaGmz, 2018-01-18:
Hello, I have a similar issue although I'm running Kinetic

As stated in the steps, after downloading the repo into my src folder:
`$ sudo nano /etc/ros/rosdep/sources.list.d/30-ubiquity.list `
and added:
`yaml https://raw.githubusercontent.com/UbiquityRobotics/rosdep/master/raspberry-pi.yaml`
Then 
`run rosdep update`
then
`cd ~/catkin_ws`
`rosdep install --from-paths src --ignore-src --rosdistro=kinetic -y`

and I get:

`ws$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Hit https://raw.githubusercontent.com/UbiquityRobotics/rosdep/master/raspberry-pi.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
Add distro "groovy"
Add distro "hydro"
Add distro "indigo"
Add distro "jade"
Add distro "kinetic"
Add distro "lunar"
Add distro "melodic"
updated cache in /home/adrianna/.ros/rosdep/sources.cache

$ rosdep install --from-paths src --ignore-src --rosdistro=kinetic -y
dpkg-query: no packages found matching libraspberrypi-dev
dpkg-query: no packages found matching libraspberrypi0
dpkg-query: no packages found matching libraspberrypi-dev
dpkg-query: no packages found matching libraspberrypi0
executing command [sudo -H apt-get install -y libraspberrypi-dev]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libraspberrypi-dev
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y libraspberrypi-dev] failed

$ sudo -H apt-get install -y libraspberrypi-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libraspberrypi-dev`

can you please help me out?

rohbotics, 2018-01-18:
What OS are you using?

If ubuntu, which image are you using?

AdriannaGmz, 2018-01-19:
Linux Ubuntu 16.04

rohbotics, 2018-01-19:
@AdriannaGmz where did you get your ubuntu install for the pi?

We have tested on both [our image](downloads.ubiquityrobotics.com) and the [Ubuntu Mate](https://ubuntu-mate.org/raspberry-pi/)  image. 

Does you install have a `/opt/vc` directory? What is the output of `ls -al /opt/vc`?

AdriannaGmz, 2018-01-28:
now is working, thank you. i had to reinstall python

rohbotics, 2018-01-28:
Good to hear that your problem is solved.

Good luck building out whatever you are using `raspicam_node` for!

closed 2018-01-28 by rohbotics

agnihotri123, 2019-07-25:
Hey `raspi cam`

drtritm, 2020-03-04:
> ur ubuntu install for the pi?

Hi, I got the same problem, but I install ROS Master on my laptop, I use ubuntu 16.04, ros kinetic, when I trying to build from source for my laptop to use motion vector. And the output of ls -al /opt/vc is no directory

rohbotics, 2020-07-01:
Hi @Ash-S 

Are you using raspbian? How did you install ROS?

If you installed from source, you will need to manually download and install that dependency.

Ash-S, 2020-07-01:
Hi @rohbotics , thanks for your reply. I figured it out while watching the compiler errors...i needed to define "add_compile_options(-std=c++11)" in src/image_common/camera_calibration_parsers/CMakefile and package build works. But there is again an issue while running launch file. 
**_"roslaunch raspicam_node camerav2_1280x960.launch"_**, Could you please check and suggest? Thanks.

... logging to /home/pi/.ros/log/27983cee-bb7c-11ea-949c-b827eb16b4c4/roslaunch-raspberrypi-1165.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raspberrypi:34295/

SUMMARY
========

PARAMETERS
 * /raspicam_node/camera_frame_id: raspicam
 * /raspicam_node/camera_id: 0
 * /raspicam_node/camera_info_url: package://raspica...
 * /raspicam_node/camera_name: camerav2_1280x960
 * /raspicam_node/enable_imv: False
 * /raspicam_node/enable_raw: False
 * /raspicam_node/framerate: 30
 * /raspicam_node/height: 960
 * /raspicam_node/private_topics: True
 * /raspicam_node/width: 1280
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    raspicam_node (raspicam_node/raspicam_node)

auto-starting new master
process[master]: started with pid [1177]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 27983cee-bb7c-11ea-949c-b827eb16b4c4
process[rosout-1]: started with pid [1190]
started core service [/rosout]
process[raspicam_node-2]: started with pid [1197]
terminate called after throwing an instance of 'std::length_error'
  what():  vector::_M_fill_insert
**[raspicam_node-2] process has died [pid 1197, exit code -6, cmd /opt/ros/kinetic/lib/raspicam_node/raspicam_node __name:=raspicam_node __log:=/home/pi/.ros/log/27983cee-bb7c-11ea-949c-b827eb16b4c4/raspicam_node-2.log].
log file: /home/pi/.ros/log/27983cee-bb7c-11ea-949c-b827eb16b4c4/raspicam_node-2*.log**

rohbotics, 2020-07-02:
Having to modify camera_calibration_parsers doesn't sound quite right, I think moving it to C++11 is causing the runtime error that you are getting now, but I am not sure.

Can you post the compiler errors you were getting?

Can you also tell me more about your OS and how you installed ROS?

Ash-S, 2020-07-02:
Hi, yes. I installed ROS (raspbian) through here: http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi

Ash-S, 2020-07-02:
Since first when i got cam package from here https://github.com/UbiquityRobotics/raspicam_node, i tried to build it but it was showing errors for camera_info_manager, so i got package "image_common" http://wiki.ros.org/image_common from source. I tried build again and got errors related to "camera_calibration_parsers", so i tried build individual package of "camera_calibration_parsers", and i got the compiling errors and fixed those like mentioned here: https://answers.ros.org/question/216842/ros-using-c-11-how-to-use-with-catkin/?comment=356182#post-id-356182

Then I build "raspicam_node" again and its done. Now but when I launch "camerav2_1280x960.launch", I face there erors:

**setting /run_id to 27983cee-bb7c-11ea-949c-b827eb16b4c4
process[rosout-1]: started with pid [1190]
started core service [/rosout]
process[raspicam_node-2]: started with pid [1197]
terminate called after throwing an instance of 'std::length_error'
what(): vector::_M_fill_insert
[raspicam_node-2] process has died [pid 1197, exit code -6, cmd /opt/ros/kinetic/lib/raspicam_node/raspicam_node __name:=raspicam_node __log:=/home/pi/.ros/log/27983cee-bb7c-11ea-949c-b827eb16b4c4/raspicam_node-2.log].
log file: /home/pi/.ros/log/27983cee-bb7c-11ea-949c-b827eb16b4c4/raspicam_node-2.log**

Please Suggest! Thanks.

Ash-S, 2020-07-02:
@rohbotics; while building 'camera_calibration_parsers' i was getting these errors.

[image]

rohbotics, 2020-07-02:
Thanks for the further info.

It seems like something is pretty broken with dependency versions, as the underlying libraries that ROS uses have gotten incompatible updates in raspbian since Kinetic came out.

Maybe you would have better luck trying to adapt those instructions to install Melodic or maybe Noetic instead?

 Alternatively you can use our Ubuntu Xenial image here: https://downloads.ubiquityrobotics.com/pi.html

Ash-S, 2020-07-02:
Thanks but Since along with Raspicam i need to use 'dynamixel-workbench' package on it, the Ubuntu Xenial image  https://downloads.ubiquityrobotics.com/pi.html would support this without any further issues, if you have tested it?
Because after a lot of hard work to resolve the issues and to get 'working ROS' on RPi3, i am able to use 'dynamixel-manager' on it...but not while i need to add Pi-Camera on it, i am facing these issues as you mentioned its of "broken dependencies"...didnt understand well but if this all work well with 'Ubuntu Xenial Image', i can think of to move on this. Please suggest. Thanks.

rohbotics, 2020-07-02:
dynamixel_workbench looks like it supports ROS Kinetic which means it should work on the Ubuntu Xenial + ROS Kinetic we have on that image. You should be able to put it in your catkin_ws and compile it.

Otherwise the thing to try would be to get a newer version of ROS installed on your Pi, because it seems Kinetic is too old to work properly on the new versions of raspbian.

Good luck with your project!

Ash-S, 2020-07-02:
Okay. thanks Rohan @rohbotics for your kind sugestions. I'd look forward it. Seems, I have 2 options now to move on:
1. prefer to use Ubuntu Xenial+ROS Kinetic
2. new version of ROS which supports well  "RPi3+dynamixel_workbench+PiCam+Radar"; any suggestions/recommendations please?
Thanks.

rohbotics, 2020-07-02:
For option 2, you might have better luck trying to compile ROS Melodic from source (instead of Kinetic), but I haven't tried it, so I can't really provide any recommendations.

Ash-S, 2020-07-06:
Thanks @rohbotics . I'd go through it and test. well, may i please know is there any specific configuration for WiFi connect RPi (in your image Ubuntu Xenial). I am trying but its failed. Thanks.

ivantomic17, 2020-07-06:
Hi, @rohbotics  I am running ROS melodic on Raspberry PI 4B (Raspbian) and I am trying to install pylon_camera. I have a problem, when I run:
rosdep install --from-paths src --ignore-src --rosdistro melodic -y -r --os=debian:buster

I get :
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
pylon_camera: No definition of [pylon] for OS [debian]

If anyone can help me i would appreciate it.