UbiquityRobotics/raspicam_node #37: How to use the node for camera calibration

ROBOTIS-Leon, 2018-01-10:
Hi, I'm Leon, an engineer from ROBOTIS
I noticed that this package came from https://github.com/fpasteau/raspicam_node which is currently not in service, and I'm very appreciated that you are continuing management of thus very important package.

By the way, I'm currently doing some works that uses this raspicam package for an official [TurtleBot3 example](http://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#turtlebot3-auto),
but to do that, I needed to do camera calibration for raspberry pi camera. The thing is, the node doesn't send raw image which was available previously.

As I know, camera calibration node needs the raw image, and as you commented in another issue, yes I need to make an extra decoder node. Plus, I need to modify the code a little to make it does service response to camera calibration node. However, to put this into official TurtleBot3 official package, I should minimize a number of packages used. 

So here is what I need.

1. Could you bring raspicam_raw_node here? or modify raspicam_node to become publishable the raw image. 

2. Could you add the service response code for camera calibration node?

If you are very busy, but also hope that those parts comes in the package, I can make it and send a PR. 

Cheers.

rohbotics, 2018-01-10:
@ROBOTIS-Leon 

Cool to hear that you are using `raspicam_node` in an official Turtlebot3 example.

The image republish node is part of `image_transport` a core ros package, so I don't think it makes sense to avoid pulling it in as a dependency, as it likely will be one anyway.

I am not sure what you mean by 'service response code for camera calibration node' could you clarify this? The package works with the calibration program's save and commit buttons right now.

If you are interested in making these changes a PR is welcome.

davecrawley, 2018-01-10:
> raspicam_node which is currently not in service, and I'm very appreciated

@ROBOTIS-Leon <https://github.com/robotis-leon>

I think it would be great if you did a PR request so you can get exactly
what you need. We will promptly review it and merge it in to the main
branch as required. In due course this will make its way on to our standard
Raspberry Pi image. You can get this image at
https://downloads.ubiquityrobotics.com if it is helpful to your future
endeavors.

Requests for what should be on the Raspberry Pi image can also be logged on
https://github.com/UbiquityRobotics/magni_robot. So if when you look at the
Raspberry Pi image you find things that you want on there let us know
through a github issue on that repository.

Cheers,

David

ROBOTIS-Leon, 2018-01-10:
@rohbotics 

No problem,
I couldn't find the set_camera_info service from the raspicam_node, but now I found it.

Thank you.

@davecrawley 

Thank you for your kindness. 
Getting advices from @rohbotics, now all the things are crystal clear. 

You can close this issue

rohbotics, 2018-01-12:
Glad to help.

Feel free to open issues if you encounter further problems.

closed 2018-01-12 by rohbotics