UbiquityRobotics/raspicam_node #38: [raspicam_node-2] process has died on Raspberry pi3, Linux raspberrypi 4.9.59-v7+

shmm91, 2018-01-13:
... logging to /home/pi/.ros/log/dc2f88fc-f841-11e7-b831-b827eb75bfae/roslaunch-raspberrypi-1075.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raspberrypi:36793/

SUMMARY
========

PARAMETERS
 * /raspicam_node/camera_frame_id: raspicam
 * /raspicam_node/camera_info_url: package://raspica...
 * /raspicam_node/exposure_mode: antishake
 * /raspicam_node/framerate: 10
 * /raspicam_node/height: 960
 * /raspicam_node/shutter_speed: 0
 * /raspicam_node/width: 1280
 * /rosdistro: kinetic
 * /rosversion: 1.12.12

NODES
  /
    raspicam_node (raspicam_node/raspicam_node)

auto-starting new master
process[master]: started with pid [1085]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to dc2f88fc-f841-11e7-b831-b827eb75bfae
process[rosout-1]: started with pid [1106]
started core service [/rosout]
process[raspicam_node-2]: started with pid [1117]
[ INFO] [1515834746.562220740]: Loading CameraInfo from package://raspicam_node/camera_info/camerav2_1280x960.yaml
[ INFO] [1515834746.899745114]: Camera component done

[ INFO] [1515834746.906983500]: Encoder component done

[ INFO] [1515834746.932041469]: camera calibration URL: package://raspicam_node/camera_info/camerav2_1280x960.yaml
[raspicam_node-2] process has died [pid 1117, exit code -11, cmd /home/pi/ros_ws/devel/lib/raspicam_node/raspicam_node __name:=raspicam_node __log:=/home/pi/.ros/log/dc2f88fc-f841-11e7-b831-b827eb75bfae/raspicam_node-2.log].
log file: /home/pi/.ros/log/dc2f88fc-f841-11e7-b831-b827eb75bfae/raspicam_node-2*.log

rohbotics, 2018-01-13:
What OS are you running on?

Did you install from source or from debs?

shmm91, 2018-01-14:
The OS I use is Raspbian Stretch with desktop.
I have installed ROS Kinetic from source, and git clone the raspicam_node in my workspace, then build it from source.
Do you know how to fix this issue? Thank you !

rohbotics, 2018-01-21:
The only thing I can think of is making sure you have all the dependencies in your source ROS install.

```
compressed_image_transport
roscpp
std_msgs
std_srvs
sensor_msgs
camera_info_manager
dynamic_reconfigure
```

davecrawley, 2018-01-21:
Consider also, if you continue to have difficulty resolving dependencies
that we have a prepared Raspberry Pi image with all the dependencies worked
out already. All you have to do is download the image and write it to a SD
card. This image is known to work although it may contain things that you
do not need.

It can be found here.

downloads.ubiquityrobotics.com

shmm91, 2018-01-26:
Thank you! I have used the ubiquityrobotics image, and it works well.
Also, the Raspbian Stretch with desktop image seems doesn't support the raspicam_node.

closed 2018-01-26 by shmm91