UbiquityRobotics/raspicam_node #40: Horizontal/Vertical flip being ignored

chrisspen, 2018-02-22:
Hi, like you, I've also been maintaining my own fork of the [raspicam_node project.](https://github.com/chrisspen/raspicam_node) To simplify maintenance, I'd like to drop my project and switch to yours. However, I'm having trouble getting all the same parameters working with your node.

I noticed that you've also implemented the vFlip and hFlip flags to enable the camera's hardware support for image flipping, but these don't seem to be accessible via launch parameters.

I have a Pi camera mounted so that it needs to be flipped both vertically and horizontally in order to appear "upright", so I'm using the launch file:

    <launch>
        <node type="raspicam_node" pkg="raspicam_node" name="raspicam_node" output="screen">
            
            <param name="framerate" type="int" value="30" />
            <param name="exposure_mode" value="antishake" />
            <param name="shutter_speed" value="0" />
            <param name="quality" type="int" value="10" />
            <param name="width" type="int" value="320" />
            <param name="height" type="int" value="240" />
            
            <param name="hFlip" type="int" value="1" />
            <param name="vFlip" type="int" value="1" />
            
            <param name="camera_frame_id" value="raspicam" />
            
        </node>
    </launch>

and when I run it with:

    roslaunch myrobot myraspicam.launch

I see the output:

    [ INFO] [1519266611.729751741]: Loading CameraInfo from package://raspicam_node/camera_info/camerav2_410x308.yaml
    [ INFO] [1519266611.869569770]: Camera component done

    [ INFO] [1519266611.876421849]: Encoder component done

    [ INFO] [1519266611.891155693]: camera calibration URL: package://raspicam_node/camera_info/camerav2_410x308.yaml
    [ INFO] [1519266611.950335288]: Camera successfully calibrated
    [ INFO] [1519266612.084681134]: Reconfigure Request: contrast 4, sharpness 0, brightness 55, saturation 0, ISO 460, exposureCompensation 0, videoStabilisation 0, vFlip 0, hFlip 0, zoom 1.00, exposure_mode antishake, awb_mode auto
    [ INFO] [1519266612.089540454]: Reconfigure done
    [ INFO] [1519266612.144103542]: Starting video capture (320, 240, 10, 30)

    [ INFO] [1519266612.145129791]: Video capture started

showing that even though I'm passing in vFlip=1 and hFlip=1, they're still both set to 0. Viewing image feed from rqt_image_view confirms the image renders in the original unflipped layout. 

Why is this? Am I not using the param tags correctly, or is this not yet supported fully? If not yet supported, I've fully implemented this in my own node, so I could submit a patch if you're accepting contributions.

rohbotics, 2018-02-22:
Thanks for opening an issue.

I would try setting the parameters as bools instead of ints. 
Additionally, see if the dynamic reconfigure system allows you to change the parameter. It may be that the dynamic stuff is somehow overwriting the rosparam.

chrisspen, 2018-02-22:
I tried changing them to bools. That had no effect.

davecrawley, 2018-02-23:
We are absolutely accepting contributions, so feel free to do a PR. @jimv
@rohan what do you think?

David

rohbotics, 2018-02-24:
@chrisspen I think this a bug with how rosparams and dynamic reconfigure are interacting (namely the default dynamic reconfigure setup is overriding the rosparams).

Does using `rosrun rqt_reconfigure rqt_reconfigure` and setting the flips cause the image to flip on your system? 

I'll look into the parameter issue.

wsung1, 2018-07-28:
Hi,
I have the almost same problem, namely, the default dynamic reconfigure setup is overriding the rosparams.
As my answer to @rohbotics's question, yes,
$ rosrun rqt_reconfigure rqt_reconfigure
works fine.
However, from the next launching, it gets back to the default, which means the default dynamic reconfigure setup is overriding.
Any update?

PR #50 (2018-10-15): Major Code Clean (Part 2)
* Use unique_ptr[] for buffers instead of malloc
* Make a pool_ptr that supports clean deletion
* More references :+1: , less pointers :-1: 
* Cleanup dynamic reconfigure. Should fix #31 and fix #40 
* Slight perf improvement by reserving vector beforehand and using std::copy
* Better use of types (less casting)

Tested as working
closed 2018-10-15 by rohbotics