UbiquityRobotics/raspicam_node #52: Grayscale images

jokla, 2018-10-15:
Hi!

Would it be possible with raspicam_node to get from the raspicam a grayscale image and publish it instead of the color one?  

Thank you!

rohbotics, 2018-10-15:
It may be possible, but probably isn't something are going to implement in the near future.

If it is reasonably simple, then its possible that we can add this feature.

jokla, 2018-10-18:
Hi!

Thank you very much for the answer!

closed 2018-10-18 by jokla

DavidRoss3, 2018-10-26:
Hi, I am a developer who is new to github and open source in general. I happen to own a pi a+ with a camera module v1, and I have some c/c++ experience and ROS exposure.

I think this issue is something that I could look into starting tomorrow. Ill try to get up to date with the source code in the mean time.

kmtm, 2022-01-24:
Wondering if anyone has implemented? I'm looking to capture images in BW to port into a CNN and would love to remove an intermediary step with cv2 if possible.