UbiquityRobotics/raspicam_node #59: Raspbian installation: "No definition of [dynamic_reconfigure] for OS [debian]"

MrGrimod, 2018-11-08:
I followed the installation guide but when I run:

`rosdep install --from-paths src --ignore-src --rosdistro=indigo -y`

I get this error:

`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
raspicam_node: No definition of [dynamic_reconfigure] for OS [debian]`

I'm using a Raspberry 3 b+ with Raspbian 9 Stretch and a Indigo ROS distro.

rohbotics, 2018-11-09:
Indigo doesn't technically support debian (or raspbian), so your going to have trouble resolving packages.

How did you install ROS?

MrGrimod, 2018-11-09:
I used this: http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Indigo%20on%20Raspberry%20Pi tut. 
But does Kinetic support Raspbian?

rohbotics, 2018-11-09:
http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi

MrGrimod, 2018-11-09:
Ok, dumb question but I can't find instructions to remove the old indigo distro..

Kiana33, 2018-11-20:
@rohbotics 
Hi, I have got a similar issue here:
`raspicam_node no definition of [camera_info_manager] for OS [debian]`

I have also tried
`sudo apt-get install ros-kinetic-camera-info-manager`
and 
`sudo apt-get install camera-info-manager`
They didn't work at all.

I'm using ROS kinetic and Raspian OS. The camera is V2. Could you help me figure that out?

samsilverman, 2020-01-02:
Im getting a similar error:

`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
raspicam_node: No definition of [diagnostic_updater] for OS [debian]`

Has this issue been addressed or has a workaround been found?

samsilverman, 2020-01-03:
Nevermind I have addressed this problem. If people feel my workaround may be helpful to the issue with camera_info_manager, I can post what I did here.

Niels-Osnabrugge, 2020-01-24:
@samsilverman I currently also get the error : 
raspicam_node: No definition of [diagnostic_updater] for OS [debian]

What did you do to fix it?

closed 2020-01-24 by cy8berpunk