UbiquityRobotics/raspicam_node #60: Error when installing "Some packages could not be installed".

CzokNorris, 2018-11-09:
By using the command from the manpage, I get: 
pi@raspberrypi:~/rosbots_catkin_ws $ sudo apt install ros-kinetic-raspicam-node

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-kinetic-raspicam-node : Depends: ros-kinetic-camera-info-manager but it is not installable
                             Depends: ros-kinetic-compressed-image-transport but it is not installable
                             Depends: ros-kinetic-dynamic-reconfigure but it is not installable
                             Depends: ros-kinetic-roscpp but it is not installable
                             Depends: ros-kinetic-sensor-msgs but it is not installable
                             Depends: ros-kinetic-std-msgs but it is not installable
                             Depends: ros-kinetic-std-srvs but it is not installable
E: Unable to correct problems, you have held broken packages.

I am using kinetic. On a raspberry Pi 3B+. Camera is connected and workst with raspistill. 
I tried installing ROS from both: 
1. This image here: https://medium.com/@rosbots/ready-to-use-image-raspbian-stretch-ros-opencv-324d6f8dcd96
2. Manually: http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi

Both approches yield the same error.
Any idea what to do?

rohbotics, 2018-11-12:
If you installed ROS from source (which it seems you did) you cannot use the debian for raspicam_node.

Zarbokk, 2019-02-06:
> If you installed ROS from source (which it seems you did) you cannot use the debian for raspicam_node.

I think it should work if you install the right dependencies by yourself I just installed it on Debian raspbian stretch. (now after 3 Days) 

But i needet to adjust one think in the Cmake file and used catkin build with a custom version of OpenCV

```
find_package(catkin REQUIRED COMPONENTS
  compressed_image_transport
  roscpp
  std_msgs
  camera_info_manager
  dynamic_reconfigure
  message_generation
)
```

you additionally have to insert/ add std_srvs as another dependencie
else you dont find std_srvs /Empty.h (at least it was for me the case)

This Link was very Helpful for installing it on Stretch: https://venelinpetkov.com/2017/11/19/how-to-install-a-raspberry-camera-node-on-ros-kinetic-raspbian-stretch/