UbiquityRobotics/raspicam_node #61: Framerate at 1/3

Zarbokk, 2018-12-02:
I am using this package now for a few weeks. But suddenly the frame rate of the published images is around 1/3 of the frame rate I use while initializing the node. 10->3.3 or 60->20 

I use v2 cam and ROS Kinetic. I already re-installed the package from source and not from source. 

any idea?

rohbotics, 2018-12-02:
Are you looking at the frame rate over the network? Running over the network can lead to pretty bad framerates, especially with wifi.

Zarbokk, 2018-12-02:
sorry totally forgot to mention that, 
both. On the py i get 1/3 and in the network (wifi) i get most 3 fps(at 1280x960), which is not a big deal. if i use the lower resolution, like 410x... i still get 1/3 so this seems to not make a difference.

sean85914, 2019-01-20:
I got the same result even if I set the master at pi itself.
I used 640*480 with 30 fps but when I 

> rostopic hz  

to compressed image, it only got about 10 hz.

Zarbokk, 2019-01-20:
Yea i just live with it. For Performance it does not matter in my experience. 
just use 90 hz if you want 30 ;) thats how i do it now

AlbertLIN95, 2019-01-28:
I use 120 hz and got 43hz. For 90 hz, I got 34. It is amazing....

rohbotics, 2019-01-28:
Maybe I should add a multiply by 3 to the code :) 

More seriously, after looking into this I have no idea what is causing the frame rate not to be correct.

AlbertLIN95, 2019-01-29:
The frame rate is simply 1/3. The wrong frame rate is due the network
issue. (Even if it is wired connection. The network increase the frame rate
lol.....)

rohbotics, 2019-02-02:
I don't have a pi for the next couple days, could someone try changing the VIDEO_FRAME_RATE_DEN to equal 1?

https://github.com/UbiquityRobotics/raspicam_node/blob/kinetic/src/raspicam_node.cpp#L83 

I think the DEN means denominator, so this is actually just dividing by 3.

commit 6329a5e4 (2019-02-04): Stop the frame rate from being divided by 3

- Video frame rate denominator is now 1 instead of 3,
- Fixes #61
PR #69 (2019-02-04): Stop the frame rate from being divided by 3
- Video frame rate denominator is now 1 instead of 3,
- Fixes #61 

Thanks to everyone who reported and confirmed this bug.
closed 2019-02-04 by rohbotics