UbiquityRobotics/raspicam_node #63: The topic names shouldn't start with raspicam_node

leclercb, 2019-01-09:
Hi,

Thank you very much for the work. I have only one problem, the topic names shouldn't start with raspicam_node. The topic names should be as minimal as possible, to allow using namespaces.

In my project I want to topics to be /robotname/image, /robotname/image/compressed...
Normally this should be as easy as adding a namespace on the raspicam node, but currently I have to manually remap the topics which is not future proof.

This is part of ROS good practices.

Thanks in advance.

rohbotics, 2019-01-10:
I think this is something we could consider for future ROS distros we release into (like melodic), but is unlikely.

But why do you say remapping topics is not future proof? It is a perfectly standard ROS feature and will not introduce any breaking changes to raspicam.

leclercb, 2019-01-10:
Maybe "not future proof" is not exactly how I should have explained it.
What I meant, is that the launch file now needs to know every topic which is published by the node and remap it. If I was using a namespace, should you add, remove, rename a topic in the future, all the topics will still be correctly mapped.

For raspicam_node, I don't understand why the topics are prefixed with raspicam_node/.
The camera name could have been an option, or nothing at all.

But I perfectly understand the problem, changing this would be a breaking change.

What do you think about adding a parameter like "prefix" with a default value "raspicam_node/" (so that we could set no prefix or set it to the camera name in the launch file.

E.g.: <param name="prefix" value="$(arg camera_name)/"/>

And then using the namespace offers a lot of flexibility.
Using namespace my_robot would produce topics like: my_robot/my_camera_name/...

rohbotics, 2019-01-12:
That makes a lot of sense, I think a camera prefix parameter with a default of "raspicam_node" makes a lot of sense.

Can you submit a PR with this change?

rohbotics, 2019-01-13:
I think the best solution is to use a private node handle for parameters, and a global one for topics.

I am not opposed to manually prefixing the topic name with the node name if the camera_prefix is not specified.

leclercb, 2019-01-15:
Using two different node handles is indeed a good idea.
I started implement the prefix option but it's not a good solution.
The problem is that by setting the topics prefix to "camera_x", we end up with:

Topics example: "/camera_x/image"
Param exaple: "/raspicam_node/the_param"

This doesn't look very intuitive.
The param should be "/camera_x/raspicam_node/the_param" imho, the camera is the main prefix, the param is private and is itself prefixed with the node name.

To achieve that, I ended up with a really simple solution.
Simply introducing a new parameter called "private_topics" (defaults to true).

If the parameter is true:
Topics example: "/raspicam_node/image"
Param example: "/raspicam_node/the_param"

If the parameter is false:
Topics example: "/image"
Param example: "/raspicam_node/the_param"

Then using a namespace on the node makes everything clear.

With namespace "camera_x" and "private_topics" false:
Topics example: "/camera_x/image"
Param example: "/camera_x/raspicam_node/the_param"

Which is probably what we want.

PR coming as soon as I have done some more tests.

leclercb, 2019-01-15:
#64 PR Created

PR #64 (2019-01-21): Add "private_topics" parameter
EDIT (@rohbotics): Fixes #63
closed 2019-01-21 by rohbotics