UbiquityRobotics/raspicam_node #65: raspicam_node stop publishing after a few seconds of launching

sethpoh, 2019-01-22:
Hi everyone,

I was trying to run the raspicam_node with a v1.3 camera attached to a Raspberry Pi 3B+ installed with the Ubiquity Robotics Raspberry Pi image for camera calibration. I tried the following launch command - 

roslaunch raspicam_node camerav2_1280x960_10fps.launch enable_raw:=true

I can see with rostopic list that it was publishing correctly for a few seconds then it stops. I tried with camerav1_1280x720.launch instead with the same behaviour. If I drop the fps in the launch file to 10fps from 90fps the publishing only stops after a minute or so. Lowering the resolution did not change this. The behaviour was repeated by killing and rerunning the launch file again without rebooting so connection with camera is still alive? 

However I can run the motion vector command with no issue of stopping -

roslaunch raspicam_node camerav2_410x308_30fps.launch enable_imv:=true

Anyone here has seen this behaviour before?

Thank you.

WLwind, 2019-01-24:
Have you solved this problem? It happens to me too. The raspicam_node always works for less than 20s, and then stops publishing images.

sethpoh, 2019-01-24:
Not really. But it seems that if I set enable_imv:=true, the publishing can carry on indefinitely.  But I will probably take a hit on the fps by keep this enabled.

Some other things I have tried to diagnose this issue are - 
1. Swap camera V1.3 to V2.1
2. Swap the Raspberry pi 3B+ for another.
3. Refreshing from the git repository and recompiling.
4. Played around with the resolution, frame rate, and enable_raw option.

The publishing still stops but it took longer at lower frame rate. 

Currently I have both enable_imv and enable_raw option enable for camera calibration. Seem to work with no issue.

rohbotics, 2019-01-27:
Ok I am able to reproduce this. If imv is disabled, then image publishing silently stops after a few seconds.

closed 2019-01-27 by rohbotics

sethpoh, 2019-01-29:
Hi @rohbotics, thanks for the help. Now the raspicam_node works reliably without needing the imv to be enabled.

PR #66 (2019-01-27): Don't connect video encoder unless imv published
Otherwise publishing crashes after a few seconds because nothing is
consuming the output of the video encoder, clogging the entire pipeline

Fixes #65