UbiquityRobotics/raspicam_node #73: standalone library for raspicam

christianrauch, 2019-04-05:
Would you mind splitting this project into a dedicated library and a thin ROS wrapper? This would make most of the code reusable by other projects.

The ROS node has quite some functionality now, that goes beyond what can be achieved with most C/C++ libraries for the Raspberry Pi camera. It would be useful for other projects to have this functionality available without ROS dependencies. This would also allow to reuse most of the code for a ROS2 node. The ROS guidelines recommend in general to export core functionality to a library and only provide a thin wrapper for the ROS node, making most code reusable.
A stand-alone library could simply interface via OpenCV images (or plain arrays) and the ROS node would provide the configuration and communication.

rohbotics, 2019-04-07:
Thanks for the suggestion. 

What kind of features are you looking for from a C/C++ library that MMAL doesn't provide? Under the hood everything raspicam_node does is using the MMAL libraries, which AFIAK are largely provided by Broadcom. 

I could see an argument for a C++ library that provides a more idiomatic interface than MMAL, but that seems like a tremendous undertaking. Especially considering documentation, maintenance, etc.

christianrauch, 2019-04-07:
I am looking for a high-level API to simply access the camera data. Large portions of the ROS node code is for settings up the camera and reading data from it. This is independent of the ROS configuration and communication and could be extracted into a library.
E.g. I am maintaining a fork of this repo at https://github.com/christianrauch/raspicam2_node as a ROS2 node. I separated the ROS2 (`RasPiCamPublisherNode.cpp`) from the camera code (`raspicam.cpp`) and you basically communicte via callbacks.
Such a library could be created to serve as common logic for a ROS1 and ROS2 node.