UbiquityRobotics/raspicam_node #74: low framerate

christianrauch, 2019-04-07:
The `camerav1_1280x720.launch` only gives me 51 Hz on `/raspicam_node/image/compressed`. Is this expected?

With `roslaunch raspicam_node camerav1_1280x720.launch enable_raw:=true` this drops to 21 Hz on `/raspicam_node/image/compressed` and 9 Hz on `/raspicam_node/image`.

How can I achieve a framerate of 90 Hz for uncompressed raw images?

rohbotics, 2019-04-13:
In my experience you can't get 90hz with raw unless you lower your resolution. With some really basic performance analysis, the code ends up spending most of its time in memcpy, because the raw images are so large.

rohbotics, 2019-04-13:
Another note: maybe you can get a little better framerates with more GPU memory in your config.txt

christianrauch, 2019-04-13:
I found a summary of [video modes](https://www.raspberrypi.org/blog/new-camera-mode-released/):

> The result is that we now have a set of mode as follows :
> - 2592×1944 1-15fps, video or stills mode, Full sensor full FOV, default stills capture
> - 1920×1080 1-30fps, video mode, 1080p30 cropped
> - 1296×972 1-42fps, video mode, 4:3 aspect binned full FOV. Used for stills preview in raspistill.
> - 1296×730 1-49fps, video mode, 16:9 aspect , binned, full FOV (width), used for 720p
> - 640×480 42.1-60fps, video mode, up to VGAp60 binned
> - 640×480 60.1-90fps, video mode, up to VGAp90 binned

So this seems to be a hardware limitations. Images at HD resolution (1280 x 720) can ony be read at upto 49 Hz and the maximum framerate of 90 Hz can only be achieved with VGA resolution (640 x 480) or lower.

Is the node capable of publishing at this framerate? I assume that if `raspivid` can read from the graphics buffer and write to disk at this framerate, the ROS node should be able to publish at this framerate too.

Since these modes are supported by `V4L2` on the Raspberry Pi, they might also just work with the generic ROS usb camera driver node.