UbiquityRobotics/raspicam_node #75: Compiling Error from Source.

sradmard, 2019-04-10:
Hi,
I am trying to compile the latest branch of the package from source under my catkin_ws, but I'm getting the following errors:

`/home/pi/catkin_ws/src/raspicam_node/src/raspicam_node.cpp:273:11: error: ‘PORT_USERDATA {aka struct MMAL_PORT_USERDATA_T}’ has no member named ‘pState’
    pData->pState.updater.update();
           ^
/home/pi/catkin_ws/src/raspicam_node/src/raspicam_node.cpp: In function ‘int main(int, char**)’:
/home/pi/catkin_ws/src/raspicam_node/src/raspicam_node.cpp:1350:12: error: base operand of ‘->’ has non-pointer type ‘RASPIVID_STATE’
   state_srv->updater.setHardwareID("raspicam");
            ^
/home/pi/catkin_ws/src/raspicam_node/src/raspicam_node.cpp:1358:29: error: base operand of ‘->’ has non-pointer type ‘RASPIVID_STATE’
         image_pub, state_srv->updater, FrequencyStatusParam(&min_freq, &max_freq, 0.1, 10), TimeStampStatusParam(0, 0.2)));
                             ^
/home/pi/catkin_ws/src/raspicam_node/src/raspicam_node.cpp:1363:27: error: base operand of ‘->’ has non-pointer type ‘RASPIVID_STATE’
         imv_pub, state_srv->updater, FrequencyStatusParam(&min_freq, &max_freq, 0.1, 10), TimeStampStatusParam(0, 0.2)));
                           ^
/home/pi/catkin_ws/src/raspicam_node/src/raspicam_node.cpp:1367:28: error: base operand of ‘->’ has non-pointer type ‘RASPIVID_STATE’
       cimage_pub, state_srv->updater, FrequencyStatusParam(&min_freq, &max_freq, 0.1, 10), TimeStampStatusParam(0, 0.2)));
                            ^
raspicam_node/CMakeFiles/raspicam_node.dir/build.make:62: recipe for target 'raspicam_node/CMakeFiles/raspicam_node.dir/src/raspicam_node.cpp.o' failed
make[2]: *** [raspicam_node/CMakeFiles/raspicam_node.dir/src/raspicam_node.cpp.o] Error 1
CMakeFiles/Makefile2:2163: recipe for target 'raspicam_node/CMakeFiles/raspicam_node.dir/all' failed
make[1]: *** [raspicam_node/CMakeFiles/raspicam_node.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`

I would appreciate any hint on how to solve this. I have ubuntu mate and ros kinetic on my Raspberry PI3 B.

rohbotics, 2019-04-13:
Looks like something that wasn't ready got merged in. For now you can revert to the latest stable code by running `git checkout 0.4.0`

julled, 2019-04-14:
I am facing the same problem

closed 2019-04-17 by rohbotics