UbiquityRobotics/raspicam_node #83: Support ROS Melodic + Ubuntu Mate Bionic 18.02

cielsys, 2019-05-22:
tldr: 
Could you provide instructions or a binary for that environment?

Full story:
I have a Raspi3 B+ with UbuntuMate 18.02 and ROS Melodic installed with working V2 cam. Part of my motivation for this environment is that I want this Raspi ROS to interact with a Nvidia Jetson Nano ROS and this is the current matching environment for the Nano. 

Following the build instructions until this step (note substituion of 'melodic' for 'kinetic)
`$ rosdep install --from-paths src --ignore-src --rosdistro=melodic -y `

which results in
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
raspicam_node: Cannot locate rosdep definition for [libraspberrypi0]
```
But I think that dependency is/should be met:

```
$ apt policy libraspberrypi0
libraspberrypi0:
  Installed: 1.20190215-1~bionic1.3
  Candidate: 1.20190215-1~bionic1.3
  Version table:
 *** 1.20190215-1~bionic1.3 990
        990 http://ppa.launchpad.net/ubuntu-pi-flavour-makers/ppa/ubuntu bionic/main armhf Packages
        100 /var/lib/dpkg/status

$ apt policy libraspberrypi-dev
libraspberrypi-dev:
  Installed: 1.20190215-1~bionic1.3
  Candidate: 1.20190215-1~bionic1.3
  Version table:
 *** 1.20190215-1~bionic1.3 990
        990 http://ppa.launchpad.net/ubuntu-pi-flavour-makers/ppa/ubuntu bionic/main armhf Packages
        100 /var/lib/dpkg/status
```
I suspect it is a simple problem involving how the dependency is expressed and resolved, but I'm not familiar with those mechanisms.

Any guidance would be welcome!
-Chris

rohbotics, 2019-05-22:
I think you skipped the step where you add the Ubiquity rosdep overlay. 

To `/etc/ros/rosdep/sources.list.d/30-ubiquity.list` add
`yaml https://raw.githubusercontent.com/UbiquityRobotics/rosdep/master/raspberry-pi.yaml`

You can also safely add `--skip-keys libraspberrypi0` to the rosdep install command.

cielsys, 2019-05-23:
Worked, thanks. Details for others below.

I did already have the yaml ref in the sources.list, so that wasn't the problem.
So I tried your suggestion 
```
$ rosdep install --from-paths src --ignore-src --rosdistro=melodic -y --skip-keys libraspberrypi0 
...
raspicam_node: Cannot locate rosdep definition for [libraspberrypi-dev]
```

so next
```
$ rosdep install --from-paths src --ignore-src --rosdistro=melodic -y --skip-keys libraspberrypi0 --skip-keys libraspberrypi-dev
```

THAT WORKED! Rosdep succeeds then catkin_make... And then the subsequent testing invocations work... then eventually crash. But I'll investigate a bit and post another issue if needed.
THANK YOU!
-Chris

closed 2019-05-23 by cielsys

rohbotics, 2019-05-23:
Hey @cielsys glad that worked out. Feel free to open another issue, I would like to see 18.04 support working. If everything works out, I would love to know! ra@ubiquityrobotics.com

vibhu19946, 2019-10-12:
I'm unable to build `raspicam_node` on Ububtu Mate 18.05, Melodic.
`catkin build` gives error

____________________________________________________________________________________________________________________________________________________
Errors     << raspicam_node:make /home/visualnav/catkin_ws/logs/raspicam_node/build.make.003.log                                                    
/opt/vc/lib/libbcm_host.so: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
make[2]: *** [/home/visualnav/catkin_ws/devel/.private/raspicam_node/lib/raspicam_node/raspicam_node] Error 1
make[1]: *** [CMakeFiles/raspicam_node.dir/all] Error 2
make: *** [all] Error 2

rohbotics, 2019-10-12:
Are you using the 64 bit version of Ubuntu Mate?

CodingSubmarine, 2020-01-15:
Hi,

I'm also trying to get the node to work on mate 18.04. I've followed the above suggestions but am getting a different error when building with catkin_make:

CMake Error at raspicam_node/CMakeLists.txt:29 (MESSAGE):
  Could not find mmal libraries

-- Configuring incomplete, errors occurred!
See also "/home/pi/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

Hope somebody can help me how to fix this issue, I'm not the very best at these things

JCPuchalt, 2020-03-02:
> Are you using the 64 bit version of Ubuntu Mate?

I also get the same error. Yes, I have the RPi 4 with 4GB RAM and the 64 bit version.

rohbotics, 2020-03-02:
For those of you running the 64bit MATE, does the camera work at all? Does `raspistill -o cam.jpg` produce an output?

JCPuchalt, 2020-03-03:
Camera does not work
`raspistill: command not found`

rohbotics, 2020-03-03:
I don't think the Ubuntu MATE team has gotten the camera working at all on their 64 bit system. If you want the camera, I think you will have to stick with 32 bit.

CodingSubmarine, 2020-03-03:
I got the node running on Pi 3B+ which is 64 bit. I ended up using a Pi image distributed by ubiquityrobotics.

I believe it's [this ](https://downloads.ubiquityrobotics.com/pi.html) one 

The camera node is already installed and worked perfectly out of the box for me.

JCPuchalt, 2020-03-04:
I got install userland, raspi-config, rpi-update... on Pi 4 Ubuntu Mate 64 bits. Camera is working (I got images and preview) with raspivid and raspistill but it runs with some warnings:
```
mmal: mmal_vc_shm_init: could not initialize vc shared memory service
mmal: mmal_vc_component_create: failed to initialise shm for 'vc.camera_info' (7:EIO)
mmal: mmal_component_create_core: could not create component 'vc.camera_info' (7)
mmal: Failed to create camera_info component
``` 
Some comments are about permissions, but I changed them and warnings are held.
```
sudo usermod -a -G video ubuntu
```

I can compile raspicam_node, with your pieces of advice and :
`adding -nostartfiles to your linker options in CMakeLists.txt` 

but after launch node, this was "has died":

```
ubuntu@ubuntu:~$ roslaunch raspicam_node camerav1_1280x720.launch
... logging to /home/ubuntu/.ros/log/fe0c4e24-5e06-11ea-99bc-dca632551d25/roslaunch-ubuntu-3261.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:42181/

SUMMARY
==========

PARAMETERS
 * /raspicam_node/camera_frame_id: raspicam
 * /raspicam_node/camera_id: 0
 * /raspicam_node/camera_info_url: package://raspica...
 * /raspicam_node/camera_name: camerav1_1280x720
 * /raspicam_node/enable_imv: False
 * /raspicam_node/enable_raw: False
 * /raspicam_node/framerate: 90
 * /raspicam_node/height: 720
 * /raspicam_node/private_topics: True
 * /raspicam_node/width: 1280
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    raspicam_node (raspicam_node/raspicam_node)

ROS_MASTER_URI=http://localhost:11311

process[raspicam_node-1]: started with pid [3276]
[raspicam_node-1] process has died [pid 3276, exit code -7, cmd /home/ubuntu/catkin_ws/devel/lib/raspicam_node/raspicam_node __name:=raspicam_node __log:=/home/ubuntu/.ros/log/fe0c4e24-5e06-11ea-99bc-dca632551d25/raspicam_node-1.log].
log file: /home/ubuntu/.ros/log/fe0c4e24-5e06-11ea-99bc-dca632551d25/raspicam_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
 
Maybe needs to be created `camera_info component` ?

besttangent, 2020-03-08:
> make[2]: *** [/home/visualnav/catkin_ws/devel/.private/raspicam_node/lib/raspicam_node/raspicam_node] Error 1

Hi,did you solve this problem?I met this problem,too.

Fan-loewe, 2020-10-05:
> Camera does not work
> `raspistill: command not found`

Hello, did you solve the problem? I also met that :)

pardhuAj, 2022-03-03:
> Hi,
>
> I'm also trying to get the node to work on mate 18.04. I've followed the above suggestions but am getting a different error when building with catkin_make:
>
> CMake Error at [...]

Hi, 
I am also facing the same issue. Were you able to figure it out?