UbiquityRobotics/raspicam_node #84: ImageView.callback_image() while trying to convert image from 'bgr8' to 'rgb8' an exception was thrown(Image is wrongly formed:height * step !=size or 960*3840 != 1843200))

leelaser, 2019-08-25:
ImageView.callback_image() while trying to convert image from 'bgr8' to 'rgb8' an exception was thrown(Image is wrongly formed:height * step !=size or 960*3840 != 1843200))

Lashhev, 2019-11-09:
I conducted a series of tests and found out that the image resolution should be with a certain aspect ratio (4: 3) or (16: 9). Try to comply with these rules.

elgarbe, 2019-12-07:
this is the video modes list for v2 camera:
1 - 1080P30 cropped (680 pixels off left/right, 692 pixels off top/bottom), up to 30fps
2 - 3240x2464 Full 4:3, up to 15fps
3 - 3240x2464 Full 4:3, up to 15fps (identical to 2)
4 - 1640x1232 binned 4:3, up 40fps
5 - 1640x922 2x2 binned 16:9 (310 px crop T/B before binning), up to 40fps
6 - 720P bin+crop (360 px L/R, 512 px T/B before binning), 40..90fps (OC: 120fps)
7 - VGA bin+crop (1000 px L/R, 752 px T/B before binning), 40..90fps (OC: 120fps)

1640x922 and 1640x1232 does not work when raw image is enabled. can you fix it?