UbiquityRobotics/raspicam_node #90: Displaying image from raspicam_node in RViz

pitosalas, 2019-11-25:
(this was also posted in answerrs.ros.org)

- Turtlebot3 from Robotis
- Ubuntu 18.04
- ROS Melodic
- Raspi Camera installed

I have the raspbery pi camera set up well enough so that when I run rqt_image_view I see the image. Oddly the topic is something like /raspicam_node/image/compressed_mouse_left . No idea what mouse_left means in this context.

Rostopic List shows:

    /raspicam_node
    /raspicam_node/camera_info
    /raspicam_node/image/compressed
    /raspicam_node/parameter_descriptions
    /raspicam_node/parameter_updates

When I specify node=/raspicam_node/image I get the closest. 
(Note that rqt_image shows the image fine and mentions a weird topic: "/raspicam_node/image/compressed_mouse_left" which doesn't work in Riz.)

The closest I get Rviz to work with the camera is   when i specify topic raspicam_node/image. 

Then in the RVIz box I see: 

    Camera:
    Status: Warn
    Image: No Image Received
    Camera Info: received
    Topic: OK

And the image area in RVIz is just black! What am I missing?

cemoke, 2020-03-17:
I have exactly same issue regarding "Oddly the topic is something like /raspicam_node/image/compressed_mouse_left . No idea what mouse_left means in this context." however, I am not able to see the image published in rqt_image_viewer. I also have the same configuration but using my robot instead of turtlebot.

Ubuntu 18.04
ROS Melodic
Raspi Camera installed

marcostrullato, 2020-06-06:
Same situation here...

MoffKalast, 2020-06-06:
From what I know RViz isn't capable of displaying CompressedImage topics. You should be able to view them using rqt_image_view, or change the boolean in the raspicam_node launch file to enable publishing raw images. Those should be viewable in RViz.

nikodemj9, 2020-07-03:
RViz should be able to show the compressed topic, it just doesn't like when there is no raw topic. 
You need to add "Image" display, fill the "Image Topic" parameter with "/raspicam_node/image" and after that change transport hint to "compressed". I believe that should work.