UbiquityRobotics/raspicam_node #91: Stopped working after upgrade

droter, 2020-01-08:
Hi,

Thanks for writing the raspicam_node.

The RPi is able to see the camera with:
raspistill -o image.jpg 

This was working before I did an update && upgrade and now the following errors are showing up.

Started with:
ubuntu@camera:~/catkin_ws/src/weeds/launch$ roslaunch raspicam_node camerav1_1280x720.launch

NODES
  /
    raspicam_node (raspicam_node/raspicam_node)

ROS_MASTER_URI=http://camera:11311/

process[raspicam_node-1]: started with pid [1954]
mmal: mmal_vc_shm_init: could not initialize vc shared memory service
mmal: mmal_vc_component_create: failed to initialise shm for 'vc.ril.camera' (7:EIO)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (7)
mmal: Failed to create camera component
[ERROR] [1578508373.260414182]: Failed to create camera component
[ERROR] [1578508373.260778921]: init_cam: Failed to create camera component
[ INFO] [1578508373.261137722]: camera calibration URL: package://raspicam_node/camera_info/camerav1_1280x720.yaml
[ INFO] [1578508373.319405435]: Camera successfully calibrated from default file
[ INFO] [1578508373.319951945]: using default calibration URL
[ INFO] [1578508373.320158559]: camera calibration URL: file:///home/ubuntu/.ros/camera_info/camerav1_1280x720.yaml
[ INFO] [1578508373.320397100]: Unable to open camera calibration file [/home/ubuntu/.ros/camera_info/camerav1_1280x720.yaml]
[ WARN] [1578508373.320558662]: Camera calibration file /home/ubuntu/.ros/camera_info/camerav1_1280x720.yaml not found.
[ INFO] [1578508373.320703506]: No device specifc calibration found
[ WARN] [1578508373.428037593]: reconfiguring, but camera_component not initialized
[FATAL] [1578508373.503209339]: Tried to start capture before camera is inited
[raspicam_node-1] process has died [pid 1954, exit code -11, cmd /opt/ros/kinetic/lib/raspicam_node/raspicam_node __name:=raspicam_node __log:=/home/ubuntu/.ros/log/6f8dd12e-d0dc-11e5-9d1d-8f1ab06ab62c/raspicam_node-1.log].
log file: /home/ubuntu/.ros/log/6f8dd12e-d0dc-11e5-9d1d-8f1ab06ab62c/raspicam_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Any ideas?

Thanks,

Matt

droter, 2020-01-09:
Permissions were the problem.

```
ubiquity$ ls -la /dev/vcsm
crw------- 1 root root 244, 0 Jul 22 13:53 /dev/vcsm
ubiquity$ ls -la /dev/vcsm-cma
crw-rw-rw- 1 root root 10, 61 Jul 22 13:53 /dev/vcsm-cma
```

Changing to:

```
ubiquity$ sudo chmod 666 /dev/vcsm
ubiquity$ ls -la /dev/vcsm                         
crw-rw-rw- 1 root root 244, 0 Jul 22 13:53 /dev/vcsm
```

Has fixed the problem.

closed 2020-01-09 by droter

droter, 2020-02-01:
The permissions were getting reset after reboot.  

I added a line to the bottom of /etc/rc.local
sudo chmod 666 /dev/vcsm

And now it seems to work.  Not sure if this is the best approach.

rohbotics, 2020-02-09:
Hey @droter 

A better solution is posted here https://github.com/UbiquityRobotics/ubiquity_main/issues/218