UbiquityRobotics/raspicam_node #93: Compilation for Ubuntu 18.04.4 LTS

pryre, 2020-02-23:
Not sure on the priority, but I have managed to get a running copy of raspicam_node on the new [Ubuntu 18.04.4 LTS](https://ubuntu.com/download/raspberry-pi) release using ROS melodic. Everything seems to be working as expected (except I haven't actually opened an image_view as I have to run headless for the time being).

The primary problem was a linking issue, similar to [this](https://github.com/raspberrypi/userland/issues/303). Compiling in `catkin_ws` gave a binary with the following linking (and some strange compiler warnings):
```
$ ldd raspicam_node | grep mmal
	libmmal.so => /opt/vc/lib/libmmal.so (0xb6b82000)
	libmmal_core.so => /opt/vc/lib/libmmal_core.so (0xb6b66000)
	libmmal_util.so => /opt/vc/lib/libmmal_util.so (0xb6b48000)
```

**Adding the following line to `CMakeLists.txt` fixed this issue (and removed the warnings):**
```
SET( CMAKE_EXE_LINKER_FLAGS "-Wl,--no-as-needed" )
```

Resulting binary was is linked as:
```
$ ldd raspicam_node | grep mmal
	libmmal.so => /opt/vc/lib/libmmal.so (0xb53ed000)
	libmmal_core.so => /opt/vc/lib/libmmal_core.so (0xb53d1000)
	libmmal_util.so => /opt/vc/lib/libmmal_util.so (0xb53b3000)
	libmmal_vc_client.so => /opt/vc/lib/libmmal_vc_client.so (0xb5397000)
```

The actual issue that is referenced above is a bit complicated for me to dive into at the moment, but this fix seems to work temporarily at least. Hopefully it helps anyone passing by.

---

Additional info: I would recommend reading [this page](https://wiki.ubuntu.com/ARM/RaspberryPi#Raspberry_Pi_packages) for getting the native Pi packages set up. I also had to add the following to `/boot/firmware/config.txt` to get the Pi detected (adding it to `usercfg.txt` did no seem to work for both RPi2 and RPi4):
```
start_x=1
gpu_mem=128
```
The result was as detected shows:
```
$ vcgencmd get_camera
    supported=1 detected=1
```

rohbotics, 2020-02-23:
@pryre Thanks for opening this issue.

Seems like the link time optimizer is doing something incorrectly. Can you make a PR with the change that made it work?

Thanks!

GuillermoHerreraF, 2020-03-05:
Hey @pryre! I've got ROS Melodic running on my RPI 4 (under Raspbian Buster). I managed to have a copy of the raspicam_node running, however when I run the tutorial.launch and afterwards I run 'rosrun rqt_image_view rqt_image_view' there is nothing on the display... any idea what it might be?

bltDefender, 2020-04-20:
@pryre @GuillermoHerreraF how did you manage to get it running under 18.04.4? I cannot build cause libraspberrypi0 is missing. Any hint welcome, I'm stuck.

ChrisTayl2018, 2020-04-28:
I am having problems building this from src on Ubuntu Server 16.04 as well as it is unable to locate libraspberrypi-dev or libraspberrypi0. Also looking for any suggestions?

GuillermoHerreraF, 2020-04-28:
@bltDefender I changed to the usb_cam node. I could not manage to make the raspicam_node work with my system configuration. http://wiki.ros.org/usb_cam