UbiquityRobotics/raspicam_node #98: Ubuntu 18.04.4 with melodic on Pi4 - libraspberrypi0 not found

bltDefender, 2020-04-19:
Hi, was trying to make the node on my pi4 but got this error:
```
E: Unable to locate package libraspberrypi0
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y libraspberrypi0] failed
```
see screenshot attached to show rosdep source is set correctly. rosdep update ran successfully.
what am I missing?

[image]

davecrawley, 2020-04-19:
Alan - you managed to get all this working on melodic under the pi 4 right?
Can you shed some light on this one? See the github thread.

bltDefender, 2020-04-20:
Don't know about the consequences yet, but this is how I brute forced the installation of missed lib.
Libraspberrypi0 conflicts with libgles2-mesa-dev, mesa-common-dev and libegl1-mesa-dev. I force kicked them out via
`sudo dpkg - r --force-depends libgles2-mesa-dev mesa-common-dev libegl1-mesa-dev`
 I found out there is additional ppa sources being available 
`sudo add-apt-repository ppa:ubuntu-raspi2/ppa`
After apt update you can add libraspverrypi-bin and Libraspberrypi-dev.

bltDefender, 2020-04-20:
There is another way to install Libraspberrypi0

closed 2020-04-20 by bltDefender