UbiquityRobotics/raspicam_node #100: branch for melodic?

robert-meiner, 2020-06-19:
Is there a simple set of changes to make for installation on melodic?

thanks, Mark

rohbotics, 2020-06-23:
I could make one, but what melodic specific changes do you want to make?

robert-meiner, 2020-06-24:
I mean whether there are a simple set of changes to the kinetic package that will make it work with melodic.
Thanks, Mark

MoffKalast, 2020-06-24:
Well it's possible that there aren't any needed. Have you tried compiling from source on melodic? Does it fail with any errors?

robert-meiner, 2020-06-24:
Yes, I did try it.  Hence my question. 
Thanks, Mark

MoffKalast, 2020-06-24:
And what were the exact errors you got? We can't exactly fix something if we don't know what's broken.

Btw, what kind of Melodic installation is this exactly? Compiled from source on Ubuntu 16 or from a package repo on Ubuntu 18? Since there isn't any Ubiquity image for 18 right now with preinstalled Pi dependencies I would imagine all the related required libraries to the pi camera are missing by default. I assume you've installed those? Does raspivid work?

robert-meiner, 2020-06-24:
Thank you for the notes.  I am using a raspberry pi 4b+ for this part of the robot.  All the dependencies are there and everything on the pi works fine. I'm keeping it lean by using Buster.  I tried installing your kinetic node last week with catkin tools and it just immediately stopped during the build.  I didn't document the error messages, since I didn't want to spend a lot of time trying to make this work. In the meantime, I've built a USB cam workaround, which seems to be working.  

Going forward, I'll look into this page once in a while to see whether you have a build or documentation for Melodic, since it would be nice to use the built-in raspberry cam-- and I would imagine that there are a lot of melodic users out there.

I'll close the issue now, but check in once in a while.

Thanks,Mark

Best Regards,
Mark

closed 2020-06-24 by robert-meiner

MoffKalast, 2020-06-24:
> would imagine that there are a lot of melodic users out there

Citation needed :)

Kinetic is still the quite popular right now and most people will be moving directly to Noetic.