UbiquityRobotics/raspicam_node #102: Melodic?

pitosalas, 2020-08-15:
Will `sudo apt install ros-kinetic-raspicam-node` work correctly with melodic?

karsjen, 2020-09-14:
i followed the build instructions in the readme section (using the kinetic branch), typing melodic instead of kinetic when running rosdep (rosdep install --from-paths src --ignore-src --rosdistro=melodic -y) and it is working correctly. I am running it on a raspberry pi 4.

adi3, 2020-09-16:
I can confirm that the node builds and runs on melodic if we skip the --rosdistro=kinetic option. @rohbotics will the maintainers consider accepting a pull request for a melodic branch on this repo? It doesn't seem like too much work (just need to update README and put in a request at ros index), so I can take this up. Let me know

rohbotics, 2020-09-16:
Hi sorry for not following this issue. I think a README change would be great!

What OS are you guys running Melodic on? The reason I have been hesitant on saying we have melodic support is that IIRC the Ubuntu Mate images didn't have mmal installed so raspicam wouldn't even build. 

As for a debian release, it's not really possible through the ROS buildfarm because of the requirement of having mmal (a raspberry pi specific thing) installed on the build machine. We (Ubiquity Robotics) don't have a melodic buildfarm or apt repo, so I could try to make a one-off package but that's about it.. We are planning on skipping directly to Noetic in a few months.

adi3, 2020-09-17:
@rohbotics I'm running Ubuntu Mate based on Bionic (18.04), but it would likely work on vanilla Ubuntu Bionic as well. I was able to build and get the package working without any modifications to code.
Quick point to clarify - how is it that you are able to provide raspicam on kinetic via the ROS buildfarm, but cannot do the same for melodic? I'd imagine we can take the same process to make a debian release available for melodic as well.

karsjen, 2020-09-17:
@rohbotics i am running it on raspbian buster. For me, building the package was also without any issues.

rohbotics, 2020-09-21:
@adi3 We don't build on the ROS buildfarm, we build on our own buildfarm for kinetic. We don't have a melodic buildfarm, but we do have a noetic one in progress.

aladdinmck, 2021-04-20:
I've tried running '$ rosdep install --from-paths src --ignore-src --rosdistro=melodic -y'
and 'rosdep install --from-paths src --ignore-src -y' (without te --rosdistro=kinetic option) and it's not working. 
E: Unable to locate package libraspberrypi-dev
I'm running Ubuntu 18.04 on RaspberryPi 3

anfederman, 2021-04-20:
Have you tried a standard apt install ros-melodic-desktop-full????

That should work depending on whixh version of Ubuntu you have on your Pi.