UbiquityRobotics/raspicam_node #105: Compilation problem on Ubuntu 18.04.5 LTS, mmal libraries.

altineller, 2020-10-14:
Hello,

I am Ubuntu 18.04.5 LTS (64bit) on a Paspberry Pi4, and I am unable to compile raspicam_node
I get the following error:

`
CMake Error at raspicam_node/CMakeLists.txt:30 (MESSAGE):
  Could not find mmal libraries`

Any ideas on how to fix this?

Best Regards,
C.

anfederman, 2020-10-16:
Sorry to say this, but currently raspicam won't work on Ubuntu 18.04 because no one has figured out to to compile the MMAL libraries. I have been successful using the usb_cam node to see images from the raspicam, but without MMAL, fiducials won't work either. We have no work around yet, so unless you want to go back to a 16.04 image you won't have raspicam_node.

This is so frustrating as everyone was looking forward to 64bit on the Pi 4, but on the bright side, it is even worse on the Jetson Nano.

altineller, 2020-10-19:
Hello @anfederman,

I managed to get it working, but using docker under raspian, where the docker contains ubuntu18. both 32bit. It is an ugly solution, but it works. Is there any people working on this MMAL libs? Maybe we can ask support from them.

Best regards,
C.