UbiquityRobotics/raspicam_node #106: Grey gradient image in rqt

houhaol, 2020-10-16:
Hello,

My Turttlebot3 waffle_pi was configured with Ubuntu Mate 18.04 32-bit and melodic ROS. I installed this raspicam_node as your instructions but modified a bit as this issue suggest #102 which just skip the skip the --rosdistro=kinetic option.

It built well but when i run it, no image showed in rqt.
[REMOTE PC]
`roscore`
[TURTLEBOT1]
`roslaunch turtlebot3_bringup turtlebot3_rpicamera.launch`
or
`roslaunch raspicam_node camerav2_1280x960.launch`
[REMOTE PC]
`roslaunch turtlebot3_bringup turtlebot3_remote.launch`
[REMOTE PC]
`rqt_image_view`
Does not show any camera input (shows some gradient)
[image]

Can you help about this issue? Thanks a lot.

anfederman, 2020-10-16:
See if topics are echoed

Rostopic list

Look for camera topics

Then

rostopic echo /camera/image_raw*
  *Not sure the exact name

If you see data, rqt-image view should list these, and you select them. If no data, then you have either a network or loading issue. If there are network issues you will see no data in other sensor topics.

Good luck.

houhaol, 2020-10-17:
> See if topics are echoed
>
> Rostopic list
>
> Look for camera topics
>
> Then
>
> rostopic echo /camera/image_raw*
> *Not sure the exact name
>
> If you see data, rqt-image view should list these, and you select [...]

Hi,
Thanks for your reply.

I listed all published topics using `rostopic list`. They are already there, on my remote PC
[image]

Then i printed out the data using `echo`, it works. But still can not visualize in rqt_view_image.
------------------------------------------------------
I tried these on my **turtlebot3 solely**, put aside my remote pc. The **camera works** on rqt and rviz. 
On bot
`roslaunch raspicam_node camerav2_1280x960.launch
rqt_image_view'

would it be a loading issue or related to remote control?

houhaol, 2020-10-17:
I failed using a WSL system for this. I successfully made it work on my Linux remote PC now. But any idea to fix this issue on wsl.

closed 2020-12-30 by houhaol

Lmixq, 2022-08-06:
> 

I have the same problem. I listed the published topics and echo them. They are there. But I can't view on rviz and rqt by WSL and XLaunch. Have you solved it?

anfederman, 2022-08-07:
Make sure both ROS_IP and ROS_MASTER_URI are expressed as IP numbers, not
named.

Example on workstation:

   export  ROS_IP=10.0.0.24,

   export  ROS_MASTER_URI=http://10.0.0.105:11311