UbiquityRobotics/raspicam_node #107: Proper syntax for camera_info_url paramaeter

pitosalas, 2020-10-26:
I am using this in my .launch file

    <param name="camera_info_url" value="package://gpg_bran/camera_info/camerav2_1280x960.yaml"/>

And yet I am getting this error message:

[ INFO] [1603729230.468427408]: Unable to open camera calibration file [/home/pi/.ros/camera_info/camerav2_1280x960.yaml]
[ WARN] [1603729230.468570167]: Camera calibration file /home/pi/.ros/camera_info/camerav2_1280x960.yaml not found.

Can you clarify how the value= for this param is interpreted? Thanks!

MoffKalast, 2020-10-26:
Yeah I think I've had a similar issue in the past, something isn't set up right for the place the package:// syntax resolution. The file:/// syntax does work but you have to specify the entire path.

pitosalas, 2020-10-26:
Thanks. I wish there was a tad more documentation. For example the node looks for a "default" camera_info file and a "specific" one. Not sure why. Thanks though I've gotten it to work.

Pito Salas
Faculty, Computer Science
Brandeis University