UbiquityRobotics/raspicam_node #110: how to compile raspicam_node source code on ubuntu 18.04 melodic ?

Shiyu-Liu, 2021-01-06:
Hi, I saw several related issues published but couldn't find out any useful solution. 
I tried to compile raspicam_node from source on **ubuntu mate 18.04 (Melodic)**  with **Raspberry Pi 3b+**.
Following the instructions, I did **$ rosdep install --from-paths src --ignore-src --rosdistro=melodic -y**, but it returned an error:
**E: Unable to locate package libraspberrypi-dev 
ERROR: the following rosdeps failed to install 
   apt: command [sudo -H apt-get install -y libraspberrypi-dev] failed**
However, I did update the rosdep adding the yaml link in the sources list. 

I noticed that in one of the issues, someone says the source compilation is working with melodic but I don't know how and why it's not the case for me. Hope that someone could help in this issue. Thanks a lot!

M-Fatih-Ozkan, 2021-01-14:
I got same issue. Dont't know that to do. Can someone help us?

MoffKalast, 2021-01-14:
I think you may need to add the raspberry pi apt repository key for that get command to work.

anfederman, 2021-01-14:
Raspicam node will not compile on a Pi 4 Ubuntu 20.04 Arm64 due to mmal not working in that architecture.

It is unknown if this is true for ARM32 architecture on Pi 3/4 Ubuntu 18.04.   The first step would be to build a raspberry pi userland, and then see if raspistill works.   If raspistill works and raspi-config can enable the camera, you should be able to build rapicam_node.

M-Fatih-Ozkan, 2021-01-16:
> Raspicam node will not compile on a Pi 4 Ubuntu 20.04 Arm64 due to mmal not working in that architecture.
>
> It is unknown if this is true for ARM32 architecture on Pi 3/4 Ubuntu 18.04. The first step [...]

Actually i am using jetson nano. But having same problem What c an i do sir?

anfederman, 2021-01-17:
Sorry,  Raspicam_node won't make on a Jetson Nano, since the Arm64 architecture won't support the MMAL component used by the code.  This has to do with the word size in the version of c++.  The ROS usb_cam node will support the raspicam on the Nano.  The same issue prevents using the Rperry Pi 4 in Arm64 mode on our Magni robots. The Pi 4 or 3 works fine using Arm32.

Unfortunately, there is no solution at this time.  Aruco marker detection will work, but fiducial_slam won't.

M-Fatih-Ozkan, 2021-01-18:
Thank you. I found a solution actually. Jetson Nano users can use jetbot ros packages by dusty-nv. With jetbot_camera you are able to use rpi cam. Thx for answers

Shiyu-Liu, 2021-01-19:
> I think you may need to add the raspberry pi apt repository key for that get command to work.

Could you provide additional infos for that ? I couldn't find out how to add the correct key for the **sudo apt-get install libraspberrypi-dev** command to work. Thanks in advanced.

Shiyu-Liu, 2021-01-19:
> Raspicam node will not compile on a Pi 4 Ubuntu 20.04 Arm64 due to mmal not working in that architecture.
>
> It is unknown if this is true for ARM32 architecture on Pi 3/4 Ubuntu 18.04. The first step [...]

Thanks for the reply. Actually I am working on Arm64-architecture RP3b+ Ubuntu 18.04. Couldn't get userland to work until now. The building process of that package was stopped due to some errors. I will continue my investigation on that.

fabiocyro, 2022-03-17:
Hello, maybe it's a little bit late and you already found your solution but this will work:

First add this PPA: 

`sudo add-apt-repository ppa:ubuntu-pi-flavour-makers/ppa`

Then:
`sudo apt-get update`

Install:
`sudo apt install libraspberrypi-bin`

To test, do this on your Raspberry:
`raspistill -v -o test.jpg`

If everything works well, the camera will show up for a few seconds and it will take a picture.

Then you can proceed on installing Raspicam node from source. I will post it here all the steps, you might have done it but this might help someone else:

Move to catkin_ws/src (or type `cs`) and clone the repo:

`~/catkin_ws/src$ git clone https://github.com/UbiquityRobotics/raspicam_node.git`

Create this file:

`sudo nano /etc/ros/rosdep/sources.list.d/30-ubiquity.list`

Add this line to it:

`yaml https://raw.githubusercontent.com/UbiquityRobotics/rosdep/master/raspberry-pi.yaml`

Update the ros dependencies:
`rosdep update`

Move to workspace & install dependencies :

```
cd ~/catkin_ws (or use cw shortcut)
rosdep install --from-paths src --ignore-src --rosdistro=melodic -y
```
(this process takes quite some time usually)

Compile:

`cm`

you will get warnings but at the end will be ok

Some remarks for this guide:

The image we used for this was this one - ubuntu-18.04.5-preinstalled-server-armhf+raspi3.img.xz

http://cdimage.ubuntu.com/releases/18.04.5/release/ubuntu-18.04.5-preinstalled-server-armhf+raspi3.img.xz

This guide is from the lab where I work, so there are more things done before and after (because it was made for the turtlebot3), I took only the part concerning the raspicam and libs you want. In case it doesn't work we can try to go through the full guide.

Contact me after if this is the case after and best of luck :)

ricemilk456, 2022-04-13:
Hello,

My SBC is Raspberry Pi4B and my camera is rpi camera v2,
I use ubuntu 18.04 preinstalled server arm64 as my SBC OS,
and my ROS version is melodic.

I have successfully installed the raspicam_node package into my ros system and successfully use the rpi camera for video streaming, the following practices are for your reference and may be helpful to you.

First,
you need to build opencv yourself by referring to the [link](https://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#install-autorace-packages-1).
Select neotic version and click "Click to expand : Autorace Package Installation for an actual TurtleBot3".
During the installation process you may need to change some of the parameters according to your environment.

Second,
you need to build mmal package:
```
cd ~/catkin_ws/src
git clone https://github.com/6by9/userland.git
cd userland
git checkout 64bit_mmal
./buildme --aarch64
sudo cp -a ~/catkin_ws/src/userland/build/lib/. /opt/vc/lib/
sudo cp -r /opt/vc/. /usr
```

Now,
You can use the following command to test:
```
raspistill -o test.jpg
```
If the picture is successfully captured, it means that you have successfully installed the mmal-64bit package.

Final,
you can download raspicam_node package from source and try to compile:
```
cd ~/catkin_ws/src
git clone https://github.com/UbiquityRobotics/raspicam_node.git
cd ~/catkin_ws
catkin_make
```

I'm not sure if my approach is flawed, but the rpi camera works fine in my ROS environment, hope the above helps you.

Shiyu-Liu, 2022-04-14:
Hello,
Thanks a lot for the continuing replies. Just to follow up. Actually I solved my problem by bypassing the issue... because from some time I upgraded my system to Raspberry Pi 4B with Ubuntu Mate 20.04, on which the compilation of raspicam_node is surely not working. But there is a ros-support package [v4l2_camera/](https://index.ros.org/r/v4l2_camera/) working as a more friendly alternative. 
So I will close the issue. Forgot to do that but eventually it is so nice to have such solutions (may be useful for others).

closed 2022-04-14 by Shiyu-Liu

oscarkfpang, 2023-04-14:
> Hello, maybe it's a little bit late and you already found your solution but this will work (i hope xD):
> 
> First add this PPA:
> 
> `sudo add-apt-repository ppa:ubuntu-pi-flavour-makers/ppa`
> 
> Then: `sudo apt-get update`
> 
> Install: `sudo apt install libraspberrypi-bin`
> 
> To test, do this on your Raspberry: `raspistill -v -o test.jpg`
> 
> If everything works well, the camera will show up for a few seconds and it will take a picture.
> 
> Then you can proceed on installing Raspicam node from source. I will post it here all the steps, you might have done it but this might help someone else:
> 
> Move to catkin_ws/src (or type `cs`) and clone the repo:
> 
> `git clone https://github.com/UbiquityRobotics/raspicam_node.git`
> 
> Create this file:
> 
> `sudo nano /etc/ros/rosdep/sources.list.d/30-ubiquity.list`
> 
> Add this line to it:
> 
> `yaml https://raw.githubusercontent.com/UbiquityRobotics/rosdep/master/raspberry-pi.yaml`
> 
> Update the ros dependencies: `rosdep update`
> 
> Move to workspace & install dependencies :
> 
> ```
> cd ~/catkin_ws (or use cw shortcut)
> rosdep install --from-paths src --ignore-src --rosdistro=melodic -y
> ```
> 
> (this process takes quite some time usually)
> 
> Compile:
> 
> `cm`
> 
> you will get warnings but at the end will be ok
> 
> Some remarks for this guide:
> 
> The image we used for this was this one - ubuntu-18.04.5-preinstalled-server-armhf+raspi3.img.xz
> 
> http://cdimage.ubuntu.com/releases/18.04.5/release/ubuntu-18.04.5-preinstalled-server-armhf+raspi3.img.xz
> 
> This guide is from the lab where I work, so there are more things done before and after (because it was made for the turtlebot3), I took only the part concerning the raspicam and libs you want. In case it doesn't work we can try to go through the full guide.
> 
> Contact me after if this is the case after and best of luck :)

I can confirm that this method works in my Raspberry Pi Zero 2W with Debian Buster and ROS melodic installed. My ROS is compiled from source and the cameras I used are v2 and Zero Cam (v1). Use respectively v2 and v1 launch files for each.