UbiquityRobotics/raspicam_node #111: raspicam_node image_raw=true don't work on PI2

elgarbe, 2021-03-06:
Hi, I'm working with a raspberry pi 2 with ubuntu MATE 16.04.7 server (the only version that I could use with raspicam).
I've installed raspicam_node from this repo following the official instructions. I need to get uncompressed image for apriltag. So I enable raw image on launchfile. But image_view and apriltag_ros throw an error like
[ERROR] [1615055637.841805351]: Unable to convert 'bgr8' image for display: 'Image is wrongly formed: height * step != size  or  1231 * 4920 != 3075072'
I've tested a lot of resolution even heigh / 16 and width / 32.
The same test on a raspberry pi 3b+ works ok.

Then I found this repo:
https://github.com/dganbold/raspicam_node
It's very old, but it works ok on my raspberry pi 2.

Is there a way to get this official repo, working with raw images on pi2? Is there an explanation why it works on pi3b+ and not on pi2?

Thank

anfederman, 2021-03-07:
You probably don't have the /opt/vc   libraries.

These can be installed by a snap install of rpi-userland followed by copying to /op/vc

sudo snap install rpi-userland –edge

copy the userland/opt/vc from snap to opt/vc
  (this installs the mmal libraries)

You may need to rebuild the node in catkin.

Also for the camera to work you need

gpu_mem=128
start_x=1  in your /boot/config.txt

Good luck.

PS.  raspicam_node does not work on a Pi using ARM64, stick to ARM32 (Armhf)

anfederman, 2021-03-07:
Another thought  does the usb_cam ros node work on the pi 2?

elgarbe, 2021-03-07:
raspicam_node from ubiquity is working on my pi2 but with compressed image only. 
The problem is with RAW image enabled.
If I enable RAW image then running image_view gives the error I've posted.
If I disabled RAW image and running image_view with transport=compressed then I get images.
The problem is apriltag needs uncompressed images.
The other thing is raspicam_node from this repo: https://github.com/dganbold/raspicam_node works ok with raw images.