UbiquityRobotics/raspicam_node #112: Noetic release raspicam_node

rohbotics, 2021-03-23:


anfederman, 2021-03-23:
Raspicam_node catkin_makes and runs under Noetic if the /opt/vc  libraries
are installed and copied from Rpi_userland snap install.  Details in my
notes.

elpidiovaldez, 2021-04-13:
@anfederman Where are the notes that you referred to ?  I could really do with some help with this.  I have spent 2 long days trying to get raspicam_node to compile using raspberry pi os (Debian Buster 32 bit)/ROS Noetic, built from source.  It seems like an unending rabbit hole of unsatisfied dependencies.

anfederman, 2021-04-13:
Send your email address to ***@***.*** I will send my notes.

It took me two months to get it working.

anfederman, 2021-04-13:
On second thought, I included them in the melodic upgrade thread.

elpidiovaldez, 2021-04-13:
@anfederman I still can't find your notes - I looked under threads entitled 'Melodic ?' and all the threads mentioning 18.04. I'd still appreciate some guidance.  

After 2 days of slog, I got raspicam_node to compile.  I failed with the Ubiquity Robotics version, but succeeded with this [version](https://github.com/dganbold/raspicam_node).  It appears to be very similar - I have no idea why this version compiled.

I had to use wstool to merge:

- compressed_image_transport
- camera_info_manager
- dynamic_reconfigure
- diagnostics

with the ros basic build.

I added myself to the video group, added the rule for /dev/vchiq and edited /boot/config.txt to give gpu_mem=144.

The results:

- raspicamstill and raspivid work.
- mmal does not give errors.
- -ENOSPC error no longer appears.
--raspicam_node launches, but it fails as follows :-1: 

```
process[raspicam_node-1]: started with pid [1752]
[ INFO] [1618322548.534369394]: Loading CameraInfo from package://raspicam_node/camera_info/camera_module_v2_640x480.yaml
[ INFO] [1618322548.590843413]: camera calibration URL: package://raspicam_node/camera_info/camera_module_v2_640x480.yaml
[ INFO] [1618322548.634080180]: Camera successfully calibrated
[ INFO] [1618322549.009469786]: Reconfigure Request: contrast 0, sharpness 0, brightness 50, saturation 0, ISO 400, exposureCompensation 0, videoStabilisation 0, vFlip 0, hFlip 0, zoom 1.00, exposure_mode auto, awb_mode auto
[ WARN] [1618322549.009703734]: camera_component not initialized
^C[raspicam_node-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

I feel I am so close.  The camera is fine, but something is still wrong with the compilation of raspicamnode.

I'd love to put together a definitive guide of how to do this so everyone does not have to go through the same pain.  That is slightly complicated because I iterated so many times I do not know exactly what made the compile succeed !

anfederman, 2021-04-13:
I am curious, does raspistill work?

My notes are for Ubuntu 20.04, so I am not sure if they will help, but
please let me send them to you directly. Send a request to
***@***.*** Ubiquity Discourse hides your email address.

If the node compiles, but does not run, this is most likely a ROS
dependency missing like image transport. If you are still having trouble,
we can schedule a zoom or Google meet call.

Hey I get your frustration. It took me two months to get it working. The
late Carnegie Mellon roboticist Randy Proust used to say:  "Brick walls are
there for a reason... They're there to show us how much we really want
something."

anfederman, 2021-04-13:
I posted to ros answers.

elpidiovaldez, 2021-04-13:
@anfederman 

raspistill and raspivid work fine.  The camera is functioning.

I used 32 bit Raspberry Pi OS (the OS formerly known as Raspbian Buster) as my base.  I thought that this would let me use gpu acceleration via mmal and would be more optimised for Pi hardware.  So I think I got off lightly with mmal.  Also it was easy to compile a basic ROS installation using this [guide](https://varhowto.com/install-ros-noetic-raspberry-pi-4/).
Do you think going with Raspbian OS was a bad idea ?

I am currently working on a Pi3B, but I also intend to use the same setup on a Pi4.

The hard part was satisfying a ton of dependencies when building raspicam_node from source.  I added:

   - compressed_image_transport
    - camera_info_manager
    - dynamic_reconfigure
    - diagnostics
    - 
I would be very happy to talk to you on Zoom or Google Meet.

elpidiovaldez, 2021-04-15:
I finally got the Ubiquity Robotics version of raspicam_node working on ROS Noetic/Raspberry Pi OS. This solution should work on Pi 3 and Pi 4.  It uses the Pi GPU via mmal and image compression, so it is fast.  It has taken me 5 days of pain to achieve this.  I hope this helps someone avoid the same ! The procedure is:

Install Raspberry Pi OS on sd card.

Install ROS from source using this [guide](https://varhowto.com/install-ros-noetic-raspberry-pi-4/), **but with one change** - install the build requirements explicitly specifying the python3 versions: `sudo apt-get install -y python3-rosdep python3-rosinstall-generator`

If all goes well ROS is installed in /opt/ros/noetic and you can now delete catkin_ws
`sudo rm -r ~/catkin_ws`

To install raspicam_node (we will make a new catkin_ws, since ROS is already installed to /opt/ros/noetic).

```
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src
git clone https://github.com/UbiquityRobotics/raspicam_node.git

```
```
cd ~/catkin_ws

rosinstall_generator compressed_image_transport --rosdistro noetic --deps --wet-only --tar > compressed_image_transport-wet.rosinstall

rosinstall_generator camera_info_manager --rosdistro noetic --deps --wet-only --tar > camera_info_manager-wet.rosinstall

rosinstall_generator dynamic_reconfigure --rosdistro noetic --deps --wet-only --tar > dynamic_reconfigure-wet.rosinstall

rosinstall_generator diagnostics --rosdistro noetic --deps --wet-only --tar > diagnostics-wet.rosinstall

wstool merge -t src compressed_image_transport-wet.rosinstall
wstool merge -t src camera_info_manager-wet.rosinstall
wstool merge -t src melodic-dynamic_reconfigure-wet.rosinstall
wstool merge -t src diagnostics-wet.rosinstall

wstool update -t src 
rosdep install --from-paths src --ignore-src --rosdistro melodic -y --os=debian:buster

```
Now you need to manually edit the following lines in  ~/catkin_ws/src/raspicam_node/src/raspicam_node.cpp
Ensure that the mixed case names are converted to underscore style.

1273:   config.exposure_compensation, config.video_stabilisation, config.vFlip, config.hFlip, config.zoom,
1274:   config.exposure_mode.c_str(), config.awb_mode.c_str(), config.shutter_speed);
1301:   raspicamcontrol_set_exposure_compensation(*state.camera_component, config.exposure_compensation);
1302:   raspicamcontrol_set_video_stabilisation(*state.camera_component, config.video_stabilisation);
1304:   raspicamcontrol_set_shutter_speed(*state.camera_component, config.shutter_speed);

`sudo src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release --install-space /opt/ros/noetic -j1 -DPYTHON_EXECUTABLE=/usr/bin/python3
`

Enable the camera and reboot
```
sudo raspi_config
sudo reboot
```

Add paul to video group. This prevents the error 'failed to open vchiq instance'

`usermod -G video _your user name_
`
`usermod -a -G video timepi
`Test camera
`raspistill -o temp.jpg
`

Test raspicam_node

```
source ~/catkin_ws/devel/setup.bash
roscore &
roslaunch raspicam_node camera_module_v2_640x480_5fps_autocapture.launch  &
rostopic hz /raspicam_node/image/compressed

```
If camera calibration file reported as not found by raspicam_node, execute following command on raspberry pi:

`cp -r ~/catkin_ws/src/raspicam_node/camera_info ~/.ros`

anfederman, 2021-04-15:
Is this Arm32 or Arm64?

elpidiovaldez, 2021-04-16:
> Is this Arm32 or Arm64?

uname -m reports the OS architecture as armv7l, which is a 32 bit architecture.

anfederman, 2021-04-16:
Thanks. For practical reasons we prefer sticking with Ubuntu, since it
easier to support via apt update and rosdep update. If you ever figure out
how to support mmal on Arm64, drop us a line.

JesseCake, 2021-06-18:
Hmmm I spent the last 2 days solving this problem for a fresh install of Raspberry Pi OS lite (buster) and it seems there were some further problems beyond what was identified above. While I could switch to a 64 bit OS like Ubuntu and use the pre-built packages, we wouldn't be able to use MMAL to speed things along with the GPU, as the GPU as far as I know it is 32 bit, and it doesn't look like this problem for RPi is going to go away any time soon.

I'm sure this is a mega-dump beyond what I should be putting in here, but just in case it's useful I'll paste my notes I made along the way the last couple of days to solve compiling this library for Raspbian (Raspberry Pi OS) and ROS Noetic. It's based on info above as well as the ubiquitous instructions found on other blogs (I'll post it on my blog too) and should get you from a fresh system, all the way to running the library fully:

Installation instructions/troubleshooting for ROS Noetic on RPi 4

{{Use Raspberry Pi OS 32 bit “buster” lite image}}

{{assumed that you have already connected to network with wifi or ethernet - I'm ssh'ing into mine}}

sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu buster main" > /etc/apt/sources.list.d/ros-noetic.list'

sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654

sudo apt update

{{make sure system is fully up to date}}

sudo apt full-upgrade

{{reboot}}
sudo reboot

{{then continue}}

sudo apt-get install -y python3-rosdep python3-rosinstall-generator python-wstool python3-rosinstall build-essential cmake

sudo rosdep init

rosdep update

mkdir ~/ros_catkin_ws

cd ~/ros_catkin_ws

(wet means released packages):

rosinstall_generator ros_comm --rosdistro noetic --deps --wet-only --tar > noetic-ros_comm-wet.rosinstall

wstool init src noetic-ros_comm-wet.rosinstall

rosdep install -y --from-paths src --ignore-src --rosdistro noetic -r --os=debian:buster

======================================================
{{increase swap file size so we don’t run out of ram compiling}}

sudo dphys-swapfile swapoff
sudoedit /etc/dphys-swapfile
{{make this edit}}: CONF_SWAPSIZE=1024
{{save and exit}}
sudo dphys-swapfile setup
sudo dphys-swapfile swapon
=======================================================

sudo src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release --install-space /opt/ros/noetic -j3 -DPYTHON_EXECUTABLE=/usr/bin/python3

{{Finished main install}}

Confirm with:
source /opt/ros/noetic/setup.bash
Run roscore to see it runs

{{set up and compile raspicam node}}

{{add the repository for rosdep to understand the dependencies that are laid out in the raspicam source to compile (this is mainly regarding libraspberrypi-dev)}}

sudo nano /etc/ros/rosdep/sources.list.d/30-ubiquity.list

{{add this line}}
yaml https://raw.githubusercontent.com/UbiquityRobotics/rosdep/master/raspberry-pi.yaml

{{save and close, then run:}}

rosdep update

{{this will allow the next steps to run as rosdep will know what to source}}

{{make new catkin workspace}}
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src

git clone https://github.com/UbiquityRobotics/raspicam_node.git

cd ~/catkin_ws

{{initialize this fresh workspace:}}
wstool init src

{{then do the following}}
rosinstall_generator compressed_image_transport --rosdistro noetic --deps --wet-only --tar > compressed_image_transport-wet.rosinstall

rosinstall_generator camera_info_manager --rosdistro noetic --deps --wet-only --tar > camera_info_manager-wet.rosinstall

rosinstall_generator dynamic_reconfigure --rosdistro noetic --deps --wet-only --tar > dynamic_reconfigure-wet.rosinstall

rosinstall_generator diagnostics --rosdistro noetic --deps --wet-only --tar > diagnostics-wet.rosinstall

wstool merge -t src compressed_image_transport-wet.rosinstall
wstool merge -t src camera_info_manager-wet.rosinstall
wstool merge -t src dynamic_reconfigure-wet.rosinstall
wstool merge -t src diagnostics-wet.rosinstall

wstool update -t src
rosdep install --from-paths src --ignore-src --rosdistro noetic -y --os=debian:buster

{{compile and install}}
sudo src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release --install-space /opt/ros/noetic -j3 -DPYTHON_EXECUTABLE=/usr/bin/python3 

{{Enable the camera in raspi_config and reboot}}

sudo raspi_config
sudo reboot

{{Add user to video group. This prevents the error 'failed to open vchiq instance'}}

usermod -G video pi
usermod -a -G video timepi Test camera

{{enable i2c bus for arducam module}}
sudo nano /boot/config.txt

{{put in the file:}}
dtparam=i2c_vc=on

{{reboot to enable}}
sudo reboot

{{test directly - should see this as a preview on a connected monitor also}}
raspistill -o temp.jpg

{{Test raspicam_node}}

source ~/catkin_ws/devel_isolated/setup.bash
roscore &
roslaunch raspicam_node camera_module_v2_640x480_5fps_autocapture.launch  &
rostopic hz /raspicam_node/image/compressed

{{If camera calibration file reported as not found by raspicam_node, execute following command on raspberry pi:}}

cp -r ~/catkin_ws/src/raspicam_node/camera_info ~/.ros

JesseCake, 2021-06-22:
Ok that was a lot to inhale, so I've blogged it instead in case it's of use for those trying to compile this great library for Raspberry Pi OS 32bit:
[(https://jessestevens.com.au/2021/06/22/compiling-and-installing-ros-noetic-and-compiling-raspicam-node-for-raspberry-pi-os-buster-for-accelerated-camera-capture/)]

zizhenli, 2021-06-29:
> Raspicam_node catkin_makes and runs under Noetic if the /opt/vc libraries are installed and copied from Rpi_userland snap install. Details in my notes.
> […](#)

@anfederman  hi, could you please send the notes to me(zizhenli904@gmail.com)  by email that you referred to? I have spent many days trying to get raspicam_node to compile using raspberry pi os ROS Ubuntu 20.04 Noetic, built from source. it always occurs problems in the picture when doing [catkin-make.], so if possible I think the note will help me a lot.
[image]

rohbotics, 2021-07-12:
Raspicam node is now released for Noetic, on images that have a working MMAL. It has been tested and works on our upcoming Ubuntu 20.04/ ROS Noetic 32bit Raspberry Pi image.

closed 2021-07-12 by rohbotics

JesseCake, 2021-07-23:
Wow thanks @rohbotics - this will help folks using 32bit Raspbian images - you're doing great work!

pitosalas, 2021-11-23:
I just found this after struggling with solving it myself - to no avail! 

Where can I find this release that works with Noetic? Or the ubiquity robotics raspy ubuntu 20.04 image that supports this? Thanks!

roaz1801, 2021-11-25:
Having the same issues as pitosalas.

MoffKalast, 2021-11-25:
Well I'm reasonably sure that this test image is capable of running the pi camera: https://forum.ubiquityrobotics.com/t/ubiquity-ros-noetic-raspberry-pi-image-beta/864

Not entirely sure how MMAL is set up on that one, but it's probably not too dissimilar than the old ones since it's still 32 bit. It's not clear if this merged branch has already made it into apt, but you know you can clone repos from github and compile them to get the latest stuff right?

klaasg, 2021-11-30:
I just installed `raspicam_node` on **Ubuntu 20.04** with **ROS Noetic**.

1. **First make sure the camera works with the `raspistill` command.**
    I followed multiple tutorials for this and don't remember everything, but one key thing was putting the following lines in `/boot/firmware/config.txt` below `[all]`
    ```
    start_x=1
    gpu_mem=512
    ```
    Test it with `raspistill -v -o test.jpg`

2. **`raspicam_node` build:**
    ```
    cd catkin_ws/src
    git clone https://github.com/UbiquityRobotics/raspicam_node
    ```
    Install libraries (I'm not sure these are all needed):
    ```
    sudo apt install libraspberrypi-dev libraspberrypi0 libpigpiod-if-dev
    ```
    At this point, `catkin_make` said it needed some packages. It's possible I forgot one, catkin_make will tell you.
    (Example: 'compressed_image_transport' -> `apt install ros-noetic-compressed-image-transport`)
    ```
    sudo apt install ros-noetic-compressed-image-transport ros-noetic-camera-info-manager ros-noetic-diagnostic-updater
    
    cd ..
    catkin_make
    ```
    If `catkin_make` takes more than a few minutes to build, it's possibly because the Pi is out of RAM.
    After waiting like half an hour I made a swapfile with [this tutorial](https://www.linuxtut.com/en/71e3874cb83ed12ec405/) and then catkin_make only took like 2 minutes.

3. **Testing the camera node**
    On your PC:
    `$ roscore`
    On the ROBOT:
    `$ roslaunch raspicam_node camerav2_1280x960.launch`
    On your PC:
    `$ rqt_image_viewer`
    at the top of the image viewer, select the raspicam_node topic.

I hope I'm not forgetting something but I lost the exact command history.

anfederman, 2021-11-30:
https://ubiquity-pi-image.sfo2.digitaloceanspaces.com/2021-08-06-focal.img.xz

I am not sure if this is a public beta release. Extract to a 32g. Some 16s
are too small.

roaz1801, 2021-12-02:
@klaasg Do you know which image you're using? Been trying to use the one for Raspberry Pi 3+ here: https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup

klaasg, 2021-12-02:
@roaz1801 The instructions I put here are instructions to build raspicam_node for a stock Ubuntu 20.04 32-bit image (from Raspberry Pi Imager) with `ros-noetic-ros-base` installed.

anfederman, 2022-02-28:
If you are building it catkin, please do a git branch and see which is current. Also check to see where ROS thinks raspicam_node is:  roscd raspicam_node.   Then see if the launch file is where ROS thinks it is. If all that checks out, maybe reinstall and catkin_make. 

ubuntu@pi-focal:~$ roscd raspicam_node/
ubuntu@pi-focal:/opt/ros/noetic/share/raspicam_node$ ls
camera_info  cmake  launch  msg  package.xml
ubuntu@pi-focal:/opt/ros/noetic/share/raspicam_node$ cd launch/
ubuntu@pi-focal:/opt/ros/noetic/share/raspicam_node/launch$ ls
camerav1_1280x720.launch        camerav2_1280x960.launch
camerav2_1280x720.launch        camerav2_1640x1232_10fps.launch
camerav2_1280x960_10fps.launch  camerav2_410x308_30fps.launch
ubuntu@pi-focal:/opt/ros/noetic/share/raspicam_node/launch$ 

if in catkin, the valid branches are still kinetic, and raspicam_flip (specific for classic magni with camera in the forward mount.

anfederman, 2022-03-01:
> @roaz1801 <https://github.com/roaz1801> I tried to follow your
> instructions , but when i execute this
>
> $ roslaunch raspicam_node camerav2_1280x960.launch
>
> i get this
> [image: image]
> <https://user-images.githubusercontent.com/92981792/156019327-559d6c27-74ba-44ea-8cfd-c23f7fa03791.png>
>
> I tried to get some informations about the raspicam_node , but it seems it
> wasnt installed proprerly
>
> rosnode info raspicam_node
> [image: image]
> <https://user-images.githubusercontent.com/92981792/156019183-2bffe17a-b2cd-4300-bb8b-6e5cbba8ff72.png>
>
> Thanks for your help

I am not sure how you built your noetic ROS.
Assuming you are on a Pi 3 or 4, most likely you need to build it in
catkin. I don't think you can get raspicam from apt.

midoyondaime, 2022-03-01:
@anfederman  Thank you for your detailed explanation .
I figured out that i have not installed some dependencies , that s why i didnt work .
But now it works fine .
PS : i am using ubuntu 20.04 on a RPi 4B
Thanks again