UbiquityRobotics/raspicam_node #113: Problem in ROS-Noetic - Failed to create camera component - connection are fine.

abdul-mannan-khan, 2021-04-07:
I am facing issues running it under ROS Noetic. Here are the errors. 

```
setting /run_id to bbbc9de4-976c-11eb-8c0e-af2ca03c1410
process[rosout-1]: started with pid [3568]
started core service [/rosout]
process[raspicam_node-2]: started with pid [3575]
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
[ERROR] [1617777886.336327072]: Failed to create camera component
[ERROR] [1617777886.339434007]: init_cam: Failed to create camera component
[ INFO] [1617777886.339700022]: camera calibration URL: package://raspicam_node/camera_info/camerav2_1280x960.yaml
[ INFO] [1617777886.382784933]: Camera successfully calibrated from default file
[ INFO] [1617777886.383029910]: using default calibration URL
[ INFO] [1617777886.383213642]: camera calibration URL: file:///home/khan/.ros/camera_info/camerav2_1280x960.yaml
[ INFO] [1617777886.383579552]: Unable to open camera calibration file [/home/khan/.ros/camera_info/camerav2_1280x960.yaml]
[ WARN] [1617777886.383751284]: Camera calibration file /home/khan/.ros/camera_info/camerav2_1280x960.yaml not found.
[ INFO] [1617777886.383913812]: No device specifc calibration found
[ WARN] [1617777886.451623871]: reconfiguring, but camera_component not initialized
[FATAL] [1617777886.477038211]: Tried to start capture before camera is inited
[raspicam_node-2] process has died [pid 3575, exit code -11, cmd /home/khan/catkin_ws/devel/lib/raspicam_node/raspicam_node __name:=raspicam_node __log:=/home/khan/.ros/log/bbbc9de4-976c-11eb-8c0e-af2ca03c1410/raspicam_node-2.log].
log file: /home/khan/.ros/log/bbbc9de4-976c-11eb-8c0e-af2ca03c1410/raspicam_node-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

I have read details given in Troubleshooting section. There is no problem in camera connection. They are fine. Could you please put some light on my problem?

I am running ROS-Noetic on Ubuntu Mate 20.04 on Raspberry Pi 4. 

Thank you.

anfederman, 2021-04-07:
First, you need to be running the arm32 version of Ubuntu 20. Raspicam is
not supported for arm64.

The libraries for the raspicam are in /opt/vc

To install you need to do a snap install of pi-userland, and then copy the
vc folder to /opt

rohbotics, 2021-04-07:
You are seeing `ENOMEM`, do you have enough GPU memory enabled in your config.txt? Try increasing it, I think a minimum of 128mb is required.

anfederman, 2021-04-07:
Also, first see if raspistill is working. This may require a desktop, but
if it isn't found or gives error messages, these need to be fixed even
before you make raspicam_node.

abdul-mannan-khan, 2021-04-08:
Thank you very much for your response. I have installed 32bit Ubuntu mate 20.04 on Raspberry Pi 4. I followed steps given in [website](https://picamera.readthedocs.io/en/release-1.13/install.html).  I read isses given in the github [repository](https://github.com/waveform80/picamera/issues/578). I could successfully install camera using the following commands

```
sudo -s
export READTHEDOCS=True
pip install picamera
```
However, I am still getting following errors 

```
khan@khan:~$ raspistill -k
mmal: Cannot read camera info, keeping the defaults for OV5647
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly
```
anfederman and rohbotics, I am very grateful to you for your kind comments.

anfederman, 2021-04-08:
So in your /boot config.txt do you have

gpu_mem=128
start_x=1

Do you have a /dev/video0 ?

Do you have group permissions to /dev/video0 ?

Finally in /opt/vc/lib. Is there the mmal components?

I know this is a difficult problem, it took me two months to get it working
properly.

Ps. I am doing this from my phone, so there are typos.

anfederman, 2021-04-08:
raspi_config is pretty useless to configure ubuntu 20.04. I am pretty sure raspicam will work, even if raspi_config says it won't is start_x=1 and gpu_mem=128

Here are my install notes for the mmal component:

?sudo apt install chrony

????------optional---- suggest not using raspi-config on ubuntu------????
 sudo dpkg -i /tmp/raspi-config_20200601_all.deb
   10  sudo raspi-config   install userland
????? may not need if already installed from 16.04 image ????

sudo snap install rpi-userland –edge

copy the userland/opt/vc from snap to opt/vc
  (this installs the mmal libraries)

pigpiod   is needed for pi_sonar

 wget https://github.com/joan2937/pigpio/archive/master.zip
 unzip master.zip 
 cd pigpio-master/
make
sudo make install

abdul-mannan-khan, 2021-04-11:
There was nothing in /opt folder. So, I followed your instructions and successfully copied vc folder from /snap/repi-userland/10/opt/.
I could see file libmmal.so in that folder and many other files like that. However, I am still getting this error

```
khan@khan:~$ raspistill -k
mmal: Cannot read camera info, keeping the defaults for OV5647
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly
```

Thank you for your insightful help. I am really hoping that we could solve this. Thank you so much.

anfederman, 2021-04-11:
the mmal component error message may occur if you don't have group
permission to /dev/video0 -or- the camera is busted, cable not seated

ls -l /dev/video0
crw-rw----+ 1 root video 81, 0 Apr 11 08:29 /dev/video0

see if 1)   /dev/video0  exists
          2)   has group = video
          3    your default user has video group permission

you could test if it is a permission problem by trying sudo raspistill -o
test.jpg
if that works you know where the issue is.

Also,  the raspicam_node may work even if raspistill doesn't
if you try:

catkin_make --pkg raspicam_node

and get no errors

try

roslaunch raspicam_node camerv2(--pick a resolution).launch  &

if no errors, you should be able to see if the camera is producing data:

rostopic echo /raspicam_node/image/compressed  (not sure of exact topic
name)

I think you are close to fixing this. I reiterate that the mmal(raspistill)
component only works on arm32,
it won't work on arm64 no matter what you try. I can sympathize with your
dilemma it took me two months of many
failed attempts to get it working.  Once you do, fiducial_slam and
fiducial_follow seem to work without a lot of drama.

abdul-mannan-khan, 2021-04-23:
Thank you so much @anfederman for your consistent support and insightful help. This problem is solved. Now, I can see my camera working. Thanks again.

closed 2021-04-23 by abdul-mannan-khan