UbiquityRobotics/raspicam_node #118: Video Latency  of 3-4 seconds and is it possible to have a raw video/images instead of compressed?

astronaut71, 2021-08-19:
Hi Im using raspberry Pi 4 B and Camera Module V2 , so Im able to compile  and run it but even the law resolution and 30fps launch file has a Video Latency of 3-4 seconds. Is it possible to reduce it and have in some how real time? Is that because is compressed video/image or? Can i display as a raw video/image, would that help to reduce the Video Latency?

Thanks

alvgaona, 2021-09-01:
Hi @astronaut71, I've tested this on a Raspberry Pi 4 with a Raspberry Pi Camera Module v2.1. I was able to run the `kinetic` branch on a ROS Noetic distro, and I had no problem whatsoever with latency. The node is publishing at ~80 Hz for a 640x480 resolution. And, by the way, you can enable a raw image topic by setting the `enable_raw:=true` parameter.

astronaut71, 2021-09-01:
can you tell me what have you done to run it without any latency?

alvgaona, 2021-09-03:
Nothing out of the ordinary @astronaut71. My set-up is Raspberry Pi 4 with a Raspberry Pi Camera Module v2. on Ubuntu Mate 20.04 with ROS Noetic. Let me correct myself on my last comment. This node only performs at ~45 Hz for a 640x480 resolution.

One question though. When you refer to latency, is this something related to low FPS? Or is it just a fixed delay?

astronaut71, 2021-09-03:
oh. Im not using Ubuntu mate. ROS melodic on Raspbian. I refer to a delay. At the delay is not fixed. Its increase with the time. FPS is fixed, its 30

alvgaona, 2021-09-07:
Why don't you try an arm64 architecture? Raspian OS is 32 bit.  Also, I believe ROS works much better with Debian distros.

anfederman, 2021-09-07:
As far as I know raspicam_node can be built and run in ARM64 Rasperry Pi 4
Ubuntu 20.04.
The steps are to start with an Ubuntu 20.04 server sd card distro,  add
wifi via netplan configuration, lubuntu desktop and ros-noetic-desktop-full.
apt install libraspberrypi-dev and -bin.   make sure gpu_mem=128, start_x=1
in /boot/firmware/config.txt

After creating a catkin_ws   git clone the raspicam_node and make it.  you
should be able to lanch the node and view the video via rqt_image_view.

if sending video over wifi, make sure it is compressed and wifi is at 5 not
2.4.

PS: raspicam node will also work on the ARM32 Ubuntu for RPi.