UbiquityRobotics/raspicam_node #119: Noetic branch/tag

alvgaona, 2021-09-01:
Hey, is it possible to create a branch for the `noetic` distro? And, also, maybe a tag off of it.

Up to this moment, the `kinetic` branch compiles and runs successfully in ROS. It's quite confusing for people looking for a node to drive the Raspberry Pi Camera in Noetic.

If you think this is doable, I offer myself to create a PR with the changes.

> Note: It'd be also quite interesting to publish it to [index.ros.org](https://index.ros.org/p/raspicam_node/github-UbiquityRobotics-raspicam_node)

anfederman, 2021-09-01:
raspicam_node does run in noetic on a raspberry pi  4/3 (Ubuntu 20.04) as
long as /opt/vc is built for ARM32.

sudo snap install rpi-userland –edge
sudo copy the userland/opt/vc from snap to *opt*/vc

(this installs the mmal libraries)  (after which  you should be able to run
raspistill, as long as your boot config.txt has

start_x=1

gpu_mem=128

Please note this will NOT work on the ARM64 version of Pi Ubuntu 20.04.

then catkin_make the raspicam_node.

alvgaona, 2021-09-03:
Thanks, @anfederman, for the quick response. In order to have the camera running with `raspivid`, `raspistill`, and so on, you just need to install `libraspberrypi-dev` in Ubuntu 20.04. This installs MMAL libraries. 

Lastly, I need to differ with you on:
> Please note this will NOT work on the ARM64 version of Pi Ubuntu 20.04.

Because I have `raspicam_node` working on ARM64 and Ubuntu 20.04.

Nevertheless, this is not what I was asking in this issue. The main reason of this issue is to create a `noetic` version of this code, which is simply creating a branch for that distro. Is this something I could do?

anfederman, 2021-09-03:
Hmm - I did get raspicam_node to run on ARM64 - it catkin made, and I was able to run rqt_image_view.  There was an c++ error
during the make.  So I am not sure if mmal is working properly.  I will have to test further.  As far as ubiquityrobotics is concerned, if mmal works on ARM64, we will eventually release our 'noetic' deb.  Please contact Rohan about creating a noetic branch, but I am not sure that it would be any different that what we have now.  send2arohan@gmail.com

closed 2021-09-03 by anfederman

pitosalas, 2021-11-25:
Sorry to come in late. I don't know who send2arohan@gmail.com is but we still are stumped trying to get raspicam_node to run on noetic on Ubuntu 20.04. Is this a solved problem? I would really really appreciate a lead or tip here. At this point I feel like we've read everything on the web about this problem and are still stuck. Thanks for your help!

anfederman, 2021-11-25:
> Sorry to come in late. I don't know who ***@***.*** is but we

I am certain it works on raspberry pi arm32. Check the forum. It is a
matter of adding a library. I think it also works on arm64, but I need to
check. I am away from my computer. I forget the name of the library off the
top of my head. Had stumped for months.

MoffKalast, 2021-11-25:
@pitosalas are you using a raspbian kernel or a stock ubuntu image from canonical? That one will likely not work.

alvgaona, 2021-11-25:
@pitosalas this already works in Ubuntu Mate 20.04 arm64 as I described above. This should not be different from Canonical's image whatsoever.

Aringan0323, 2021-11-26:
@alvgaona I'm working with @pitosalas on this project and we have been able to successfully access the camera and use the raspicam_node on arm32 but not on arm64. The Ubuntu installation we were initially using was [this one](https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#download-turtlebot3-sbc-image) from the Turtlebot3 user manual, and it don't believe it has the raspbian kernal installed by default. [This](https://zengliyang.wordpress.com/2021/01/04/raspberry-pi-4b-ubuntu-20-04-camera/) is the tutorial that I followed to enable the camera on a fresh install of Ubuntu 20.04 LTS 32bit and it worked, but it requires mmal libraries from the [UserLand](https://github.com/raspberrypi/userland) github repository which does not have 64bit support any more. If anyone has found an alternative way to access the raspberry pi camera and use raspicam_node on Ubuntu 20.04 we would appreciate some help.

anfederman, 2021-11-27:
sudo apt install libraspberrypi-bin libraspberrypi-dev

If you have a desktop then you can see if:

raspistill -o test.jpg

works.   raspicam_node should work after that.

Here are the notes for my build.  I am pretty sure pifi won't install on
arm64.

Starting from a Pi ARM64 server install add wifi by  editing the netplan:
 
15  sudo vi /etc/netplan/50-cloud-init.yaml 
   16  sudo netplan --debug apply
   17  ifconfig
   18  sudo apt update

add netools and ssh

   19  sudo apt install net-tools openssh-server
   20  sudo apt upgrade

   29  ifconfig

add desktop
   30  sudo apt-add-repository ppa:lubuntu-desktop/ppa
   31  sudo apt install lubuntu-desktop

Add camera support libraries and test:

0  sudo apt install libraspberrypi-bin libraspberrypi-dev
   41  raspistill -o test.jpg

make sure config has start_x=1 and gpu_mem=128 specified
44  cd /boot
   45  ls
   46  cd firmware/
   47  ls *.txt
   48  more config.txt 
   49  more cmdline.txt 
   50  sudo vi config.txt 
   
Install ros noetic desktop full and other ubiquity nodes from debs or git:

56  sudo apt install ros-noetic-raspicam-node
   57  sudo apt install ros-noetic-magni-robot
   60  source /opt/ros/noetic/setup.bash 
   61  roscd
   62  sudo rosdep init
   63  sudo apt install python3-rosdep python3-rosinstall python3-rosinstall-generator python3-wstool
   64  ls
   65  sudo rosdep init
70  sudo snap install rpi-userland --edge
   71  cd
   72  mkdir -p ~/catkin_ws/src
   73  cd catkin_ws/
   74  catkin_make
   75  cd src/
   76  git clone http://github.com/ubiquityrobotics/raspicam_node
   77  cd ..
   78  source ~/catkin_ws/devel/setup.bash
   79  roscd
   80  cd
   81  cd catkin_ws/
   82  catkin_make
 94  roslaunch raspicam_node camerav2_410x308_30fps.launch 
   95  history
   96  cd catkin_ws/src/
   97  git clone http://github.com/ubiquityrobotics/magni_robot
   98  git clone http://github.com/ubiquityrobotics/pi_sonar
   99  git clone http://github.com/ubiquityrobotics/oled_display
  100  git clone http://github.com/ubiquityrobotics/oled_display_node
  101  git clone http://github.com/ubiquityrobotics/ubiquity_motor
  102  cd ..
  103  ls src
  104  cd ..
  105  catkin_make
  106  cd catkin_ws/
  107  catkin_make
  108  cd src/
  109  git clone http://github.com/wjwwood/serial
 111  catkin_make
  112  cd
  113  wget https://github.com/joan2937/pigpio/archive/master.zip
  114  unzip master.zip 
  115  cd pigpio-master/
  116  make
  117  sudo make install
  118  cd
 158  git clone http://github.com/ubiquityrobotics/fiducials
  159  cd ..
  160  catkin_make --pkg fiducials
  161  sudo apt install ros-noetic-vision-msgs
  162  catkin_make --pkg fiducials
  163  rostopic list
  164  sudo apt install ros-noetic-vision-aruco-detect
  165  sudo apt install ros-noetic-aruco-detect
  166  roscd
  167  source .bashrc
  168  roscd

finally   copy  camera_info to .ros

then raspicam_node launch followed by aruco_detect.launch work

fabiocyro, 2022-03-17:
> Sorry to come in late. I don't know who [send2arohan@gmail.com](mailto:send2arohan@gmail.com) is but we still are stumped trying to get raspicam_node to run on noetic on Ubuntu 20.04. Is this a solved problem? I would really really appreciate a lead or tip here. At this point I feel like we've read everything on the web about this problem and are still stuck. Thanks for your help!

Hello, I have raspicam_node working with Ubuntu 20.04 and on a turtlebot3 also.

It works normally.

Unfortunally i'm having trouble to make it work with a package called Autorace haha