UbiquityRobotics/raspicam_node #122: Issues adding to the signing key

davidau8195, 2021-10-06:
HI, 

I am trying to install the packages for ROS Noetic for Raspicam Node. 

I read your readme file and it mentions that I need to add the repository before running apt install ros-kinetic-node. I go to the website: https://packages.ubiquityrobotics.com/

I try to run the sudo apt-key adv ... command and got an error message saying keyserver received failed: No name. 

Is there something that I am doing wrong?

anfederman, 2021-10-06:
I don't understand. If you are running Noetic on a Raspberry Pi Ubuntu
20.04, Raspicam-node needs to be built in catkin. It should work on the
beta sd card image for ARM32, and I have built it on ARM64. You cannot
install it from Debs.

ALSO the old apt keys for Xenial expired, so we need to get those
regenerated. Apt upgrade is failing for Ubuntu 16.04. I am unclear what you
are attempting to do. Installing a kinetic version of Raspicam on a noetic
running RPi?

anfederman, 2021-10-09:
> I think this is an issue discussed on another thread in Ubiquity robotics
> discourse. I will try to find the link to get the Raspicam node to compile
> I am pretty sure it now can work on both Arm32 and ARM64 once the proper
> libraries are installed. I am away from my computer, so I will give you the
> recipe tomorrow.

This is how I got raspicam_node working via catkin_make on ARM64.  This
should work for the ARM32 RPI as well,

though Idid get it working on ARM32 by a snap install of Rpi-userland and
copying the vclib files to /opt.  the key is making sure

raspistill is working before trying to install the raspicam node.

after a standard server install,  I did the following

added wifi, lubutu and assorted goodies

sudo apt install libraspberrypi-bin libraspberrypi-dev

  added  start_x=1, gpu_mem=128  to config.txt

full desktop and catkin install

git clone of raspicam_node

closed 2021-10-19 by davidau8195