UbiquityRobotics/raspicam_node #130: Issue while building raspicam_node package on ros noetic in rpi with 64 bit raspbian OS

SaailN, 2022-08-26:
I tried building the package from source on ros noetic in rpi4 with 64 bit, downloaded all dependencies and libraries but while building catkin_make it builds completely but given following error at the end:
 ```/usr/bin/ld: /opt/vc/lib/libmmal.so: error adding symbols: file in wrong format```
Also libmmal library is installed. 
[image]

Also while running launch file i am getting the following error:
```ERROR: cannot launch node of type [raspicam_node/raspicam_node]: Cannot locate node of type [raspicam_node] in package [raspicam_node]. Make sure file exists in package path and permission is set to executable (chmod +x)```

anfederman, 2022-08-26:
instructions attached to build mmal library

SaailN, 2022-08-27:
> instructions attached to build mmal library
> […](#)

@anfederman Could u tell where are the instructions?
libmmal library is already installed and its executable is present in /opt/vc/lib. 
 As per my findings( ref: https://github.com/raspberrypi/firmware/issues/1558#issuecomment-983483179) libmmal is not supported on rpi 64 bit, and libcamera is the supported solution.

anfederman, 2022-08-27:
> As per my findings( ref: raspberrypi/firmware#1558 (comment)

If everything was built, then you should be able to run raspistill. If not,
Raspicam won't work. Make sure that the vclib that gets built in snap gats
copied to /opt.  I have Raspicam working on a pi4 in both ARM32 and ARM64
Ubuntu 20.04 Noetic. It took me many months to get it to work on 64.

Please be aware that Raspicam node will be deprecated and won't work on
Ubuntu 22.

anfederman, 2022-08-27:
The instructions were attached as a odt document. Odt uses the libre office
which emulates Ms office for Linux.

SaailN, 2022-08-27:
> If everything was built, then you should be able to run raspistill. If not, Raspicam won't work. Make sure that the vclib that gets built in snap gats copied to /opt. I have Raspicam working on a [...]

I am using raspbian os 64 bit. `raspistill` wont work with 64bit version. (ref: https://forums.raspberrypi.com/viewtopic.php?t=285868)

anfederman, 2022-08-27:
I built on started with Ubuntu ARM64 server, not Raspian. I will translate my instructions to text and send the build from scratch instructions.

anfederman, 2022-08-28:
I am sending notes as an attachment in .txt format.  You should be able to
copy/paste commands to build the Image.

starting from ubuntu 20.04 SERVER image sd card!

FIRST STEPS>

First login  with a screen and keyboard mouse, forces a password change from ubuntu!

https://itsfoss.com/connect-wifi-terminal-ubuntu/   install wifi

install net-tools
update
upgrade

install ssh
sudo apt install openssh-server

connect from remote

apt install: tasksel

Next if you want to connect to a robot you need to fix the boot delay, so the 
server doesn't hang from serial input.
disable serial console on login  (see u-boot)
I haven't been able to do this on a Pi4, but if you remove the SD card from the
Pi 4, boot in Pi 3, get to u-boot and setenv bootdelay -2 and then get back to a Pi 4 serial console boot will be disabled.

ros instructions:

http://wiki.ros.org/noetic/Installation/Ubuntu

If you don't need a desktop, you can skip installing, if you need a desktop you can start with a GUI arm64 image or:

sudo apt install gnome-session gdm3

I suggest a desktop-full install, which of course means you need a desktop.

Optional, may be un-needed depending on robot and periheral hardware being interfaced:

sudo systemctl stop ***@***.*** 
sudo systemctl mask ***@***.***

OK here is the big Kuhuna - getting the raspicam to work. FIRST take some deep breaths, and put your mind at ease, this is tedious and frustrating and not for the faint of heart. SO TAKE A CLENSING BREATH, COUNT TO 10 AND BEGIN.

_______________________________________________________________________

Build ARM64 Rpi Userland,  Follow instructions, this will build the libmmal libraries and allow you to run raspstill:

1. Userland

https://gist.github.com/satmandu/c462ab301cbe09bd6e7cf4db7f626727

This is a script you download and run as sudo:

------------script starts after ---------

#!/bin/bash  -x

workdir="${HOME}/workdir"
[[ ! -d "$workdir" ]] && ( mkdir -p "$workdir" || exit 1)
[[ ! -d "$workdir"/tmp ]] && ( mkdir -p "$workdir"/tmp || exit 1)
[[ ! -d "$workdir"/output ]] && ( mkdir -p "$workdir"/output || exit 1)
echo "workdir is ${workdir}"
tmpdir=$(mktemp -d deb_XXXX -p "$workdir"/tmp)
deb_temp=${tmpdir}/deb
extract_tmp=${tmpdir}/extract
echo "tmpdir is ${tmpdir}"

rm -rf "${workdir}"/rpiuserland.compile.log
[... 106 lines cut ...]
sudo rm -rf "${workdir}"/rpi-userland/build/
sudo rm -rf "${tmpdir}"
echo "Package is at ${workdir}/output/rpiuserland_${USERLANDREV}_arm64.deb  ."
 
#------end-------script

2.  get vc library into /opt you need to execute these commands one by one:

you need to build mmal package:
cd ~/catkin_ws/src
git clone https://github.com/6by9/userland.git
cd userland
git checkout 64bit_mmal
./buildme --aarch64
sudo cp -a ~/catkin_ws/src/userland/build/lib/. /opt/vc/lib/
sudo cp -r /opt/vc/. /usr
Now,
You can use the following command to test:
raspistill -o test.jpg

3. catkin make  raspicam_node

test with roslaunch

You now should have a full ros image, but you won't have the raspicam drivers.